contiki-os/contiki #2122: why cooja-avrora-patch-wip cannot find？And cooja do not avr-raven ？

francliu, 2017-03-06:
who have cooja-avrora-patch-wip code？ If you have this code，Can you send that to liujianfei526@163.com，thankyou

dak664, 2017-03-06:
https://github.com/anadiedrichs/cooja-avrora-patch-wip

francliu, 2017-03-07:
@dak664 Thank you，I use it on cooja avr-raven mote  succed，but when I Copying code on real avr-raven 
node， it cannot  comunicate each other.can you help me?

francliu, 2017-03-07:
@dak664 
[image]
this is my result ,but the 4455 cannot cmmunicate with 4456

dak664, 2017-03-07:
You show 802.15.4 frames but the transmission depends on ipv6 being set up properly. Are you using RPL and the DAG has formed? Or peer-to-peer with the NS response to an NA?

Use wireshark to display the ipv6 protocol in those frames, for example http://www.openmote.com/blog/ieee-802-15-4-sniffer-with-openmote.html

francliu, 2017-03-17:
@dak664thanks. I already solve this problem