contiki-os/contiki #2126: example : for a learner for communicatiobn of the mote

aitsgehu, 2017-03-07:
Hey
I m new to contiki. Found it intersting so just learning it.
please provide me a basic example code for sender and receiver so that
i can establish a communication between a mote
hope for the   responses so that i can learn it

kaleemullah360, 2017-03-07:
This book is very help full for beginner why not try this

[IoT in 5 days](http://wireless.ictp.it/school_2015/book/book.pdf)

You can start from 

> 3.4. Start Instant Contiki

Instant contiki is prebuild fully configured package, just run and play arround examples.

aitsgehu, 2017-03-07:
already using Instant contiki and d inbuild examples are there but i m not able to communicate d motes as in the examples there z no sink program or receiver program so just seeking for an example so that i can able to communicate my motes

g-oikonomou, 2017-03-23:
https://github.com/contiki-os/contiki/wiki#Help

closed 2017-03-23 by g-oikonomou