contiki-os/contiki #2127: Correct IPv6 Address for MQTT Demo

paterson, 2017-03-07:
Hi there, 

I'm looking to create a MQTT connection between an NRF52 device and a broker running locally on my machine using the mqtt-demo. I am a little confused about what I should set as my MQTT_DEMO_BROKER_IP_ADDR. I tried setting it to the IPv6 address of my machine, and have also tried the IPv6 of bt0 listed in ifconfig, but neither seems to work. I can ping the device correctly using the bt0 address, but can't seem to get it to connect with my broker. For reference, I am using an Ubuntu VM and Mosquitto. Any guidance on this would be greatly appreciated.

Any thoughts? Thanks!

g-oikonomou, 2017-03-07:
What setup are you using for your gateway? border router? 6lbr?

The mqtt client plays fine with mosquitto, so I'd check mosquitto is actually listening on the interface that you are trying to connect to. I'd also make sure there is no firewall running :)

paterson, 2017-03-07:
Hi @g-oikonomou, thanks for getting back to me so quickly! 

I think it is 6lbr but this is the area I am a little unsure about. I followed this guide and configured this: https://github.com/contiki-os/contiki/blob/master/platform/nrf52dk/README-BLE-6LoWPAN.md

Is there anything else I need to configure?

Many thanks!
Niall

g-oikonomou, 2017-03-07:
Ah BLE. OK this is one thing I'm totally unfamiliar of.

So this goes over some BLE interface that's built-in your Linux box? What's the IPv6 address of your bt0? Can you ping using the non link-local address?

paterson, 2017-03-07:
Yes. The IPv6 address is as follows:

inet6 addr: fe80::a65e:60ff:febc:ee18/64 Scope:Link

There's no other addresses that one interestingly.

Thanks so much for your help!

g-oikonomou, 2017-03-07:
I see, link-local only. In that case, make sure mosquitto is listening on bt0 and configure `MQTT_DEMO_BROKER_IP_ADDR` to be the address of bt0. I'd run wireshark to see where things go wrong.

In the longer term, you probably want to distribute a routable v6 address, as per the respective section of the guide that you linked.