contiki-os/contiki #2130: How to make a connection between 2 devices CC2650 using bluetooth?

kstheart, 2017-03-08:
I used Contiki to develop cc2650, I want to use bluetooth to connect cc2650, but i can't find useful thing in Contiki.  I just found BLE beacon in the cc26xx-demo .

Should  I  use some open source bluetooth stack in Contiki  ?

chenek, 2017-03-10:
As I know, there is no enough FLASH size on CC2650 to run both Contiki OS and full BLE Stack. This means you can only enable BLE beacon within Conitki for CC2650.

g-oikonomou, 2017-03-23:
Contiki does not feature a BLE stack at the moment (other than some work on IPv6 over BLE for the nrf52dk platform). So you'd need a fair bit of work to make this work.

closed 2017-03-23 by g-oikonomou

tsparber, 2017-03-27:
Maybe this PR https://github.com/contiki-os/contiki/pull/1936 of @spoerk is of intereset.