contiki-os/contiki #2136: Cooja MspCLI execute the "start" and "stop" commands exception.

yulincoder, 2017-03-14:
Hello,
When i type the 'start' command in MspCLI, the simulation did not start.

[image]

When the simulation is already running then i type the 'stop' command, the information will been given that "CPU is not running
command 'stop' failed with error code 1"

[image]

I use sky mote and main.exe complied by tinyos.

Thank you.

joakimeriksson, 2017-03-14:
Some of the CLI commands are only useful when MSPSim is running “standalone” and not
when integrated into another framework such as COOJA. In COOJA the simulator controls
all the running instances of MSPSim so there is no point trying to stop or start one single node.

[10qtpj 4a49asjpp7j8ey]<https://cloud.githubusercontent.com/assets/10429180/23882956/dfdaea48-089e-11e7-8ec4-84a9464b5997.png>

[qq 20170314102048]<https://cloud.githubusercontent.com/assets/10429180/23883115/f25a65bc-089f-11e7-9c4c-f24396e01f9e.png>

yulincoder, 2017-03-17:
Thanks very much.

closed 2017-03-17 by yulincoder