contiki-os/contiki #2137: rpl-border-router + tunslip - need help adding a route to connect to Internet

electronut, 2017-03-14:
I have the following setup:

1. Raspberry Pi 3 + CC1310 Launchpad running _rpl-border-router_
2. CC1310 launchpad running _http-socket_ example

On the pi I have:

> $sudo ./tunslip6 fd00:1/64 -s /dev/ttyACM0

I open `[fd00::212:4b00:e07:c1f5]` in a browser on the pi and I can see my other launchpad at `fd00::212:4b00:e07:d059`.

On the pi, I have assigned an IP address to _tun0_ as follows:

> $sudo ifconfig tun0 add 2001:470:6d:3fd::2/64 

Here's my *ifconfig*:

> $ ifconfig
> eth0      Link encap:Ethernet  HWaddr b8:27:eb:fd:f9:6f  
>           inet addr:192.168.3.184  Bcast:192.168.3.255  Mask:255.255.252.0
>           inet6 addr: 2001:470:6d:3fd::1/64 Scope:Global
>           inet6 addr: fe80::fa5c:d96c:8152:622b/64 Scope:Link
>           UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
>           RX packets:5739134 errors:0 dropped:9086 overruns:0 frame:0
>           TX packets:1422403 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1000 
>           RX bytes:1847177698 (1.7 GiB)  TX bytes:825127073 (786.9 MiB)
> 
> he-ipv6   Link encap:IPv6-in-IPv4  
>           inet6 addr: fe80::c0a8:3b8/64 Scope:Link
>           inet6 addr: 2001:470:6c:3fd::2/64 Scope:Global
>           UP POINTOPOINT RUNNING NOARP  MTU:1480  Metric:1
>           RX packets:1484198 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:802057 errors:48 dropped:0 overruns:0 carrier:48
>           collisions:0 txqueuelen:1 
>           RX bytes:804144936 (766.8 MiB)  TX bytes:142257374 (135.6 MiB)
> 
> lo        Link encap:Local Loopback  
>           inet addr:127.0.0.1  Mask:255.0.0.0
>           inet6 addr: ::1/128 Scope:Host
>           UP LOOPBACK RUNNING  MTU:65536  Metric:1
>           RX packets:225 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:225 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:1 
>           RX bytes:19328 (18.8 KiB)  TX bytes:19328 (18.8 KiB)
> 
> tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
>           inet addr:127.0.1.1  P-t-P:127.0.1.1  Mask:255.255.255.255
>           inet6 addr: fd00::1/64 Scope:Global
>           inet6 addr: fe80::1/64 Scope:Link
>           inet6 addr: 2001:470:6d:3fd::2/64 Scope:Global
>           UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
>           RX packets:139 errors:0 dropped:0 overruns:0 frame:0
>           TX packets:172 errors:0 dropped:0 overruns:0 carrier:0
>           collisions:0 txqueuelen:500 
>           RX bytes:12287 (11.9 KiB)  TX bytes:16591 (16.2 KiB)

For IPv6, I have set up a tunnel via _Hurricane Electric_ and I can ping sites like _ipv6.google.com_ fine.

I have been using the site below to test *ping6* externally:

http://www.subnetonline.com/pages/ipv6-network-tools/online-ipv6-ping.php

I am able to reach _tun0_ from outside at _2001:470:6d:3fd::2_.

But my _http-socket_ is still not working. I don't think it's able to reach outside the local network.

I am guessing that I need add a few *routes* to make this happen. Could you please help me to connect the node with the Internet?

Thanks!

mcr, 2017-03-14:
Mahesh Venkitachalam <notifications@github.com> wrote:
    > 1 Raspberry Pi 3 + CC1310 Launchpad running rpl-border-router
    > 2 CC1310 launchpad running http-socket example

okay.

    > On the pi I have:

    > $sudo ./tunslip6 fd00:1/64 -s /dev/ttyACM0

Is there a typo here?  fd00::1?

    > I open [fd00::212:4b00:e07:c1f5] in a browser on the pi and I can see
    > my other launchpad at fd00::212:4b00:e07:d059.

    > On the pi, I have assigned an IP address to tun0 as follows:

    > $sudo ifconfig tun0 add 2001:470:6d:3fd::2/64

okay, but if you are using ULA fd00: addresses on your RPL, that won't be
announced to the world.  You need to use the /64 that HE gave you.
It seems like you are expecting some kind of NAT66 behaviour.

    > tun0 Link encap:UNSPEC HWaddr
    > 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
    > inet addr:127.0.1.1 P-t-P:127.0.1.1 Mask:255.255.255.255
    > inet6 addr: fd00::1/64 Scope:Global
    > inet6 addr: fe80::1/64 Scope:Link
    > inet6 addr: 2001:470:6d:3fd::2/64 Scope:Global

So, you have added the /64 there.  You could use a different IP, although it
shouldn't matter.  I think you should be running:

   > sudo ./tunslip6 2001:470:5d:3fd::1/64 -s /dev/ttyACM0

and not using ULAs at all.

--
]               Never tell me the odds!                 | ipv6 mesh networks [
]   Michael Richardson, Sandelman Software Works        | network architect  [
]     mcr@sandelman.ca  http://www.sandelman.ca/        |   ruby on rails    [

electronut, 2017-03-15:
@mcr thank you for responding.

I used _fd00::1_ because I heard that the previous prefix of _aaaa_ has been changed. I am just adapting instructions from link below:

http://anrg.usc.edu/contiki/index.php/RPL_Border_Router

I was thinking that the nodes using _fd00_ prefix will talk to _tun0_, which will then talk to the internet. (I am a novice at IPv6 and networking in general.)

mcr, 2017-03-15:
Mahesh Venkitachalam <notifications@github.com> wrote:
    > I used fd00::1 because I heard that the previous prefix of aaaa has
    > been changed. I am just adapting instructions from link below:

    > http://anrg.usc.edu/contiki/index.php/RPL_Border_Router

    > I was thinking that the nodes using fd00 prefix will talk to tun0,
    > which will then talk to the internet. (I am a novice at IPv6 and
    > networking in general.)

The Internet doesn't know about your fd00: address (nor would it know about
your aaaa: one).  IPv6 is not IPv4+NAT.   You *will* need to learn about
routing if you intend to work with RPL.

--
]               Never tell me the odds!                 | ipv6 mesh networks [
]   Michael Richardson, Sandelman Software Works        | network architect  [
]     mcr@sandelman.ca  http://www.sandelman.ca/        |   ruby on rails    [