contiki-os/contiki #2141: Unable to cause Fragmentation

Shridharfly, 2017-03-15:
I want to force fragmentation. Irrespective of whether  SICSLOWPAN_CONF_FRAG is enabled, the control reaches this line https://github.com/contiki-os/contiki/blob/3f4436bac9a9f6da0df188372d4374693eab8a52/core/net/ipv6/sicslowpan.c#L1479.  

I have experimented with various values for MAX_PAYLOAD_LEN https://github.com/contiki-os/contiki/blob/94854177e2837a6dec0e19624887165aa7cec7d5/examples/ipv6/rpl-udp/udp-client.c#L61 but I could not force fragmentation.

I noticed that large packets are not delivered, but are not fragmented.

What is the minimum MAX_PAYLOAD_LEN at which fragmentation should happen?
How do I make fragmentation happen?

BigMikes, 2017-03-15:
Probably you should increase also the size of the buffer content here:
`sprintf(buf, "Hello %d from the client", seq_id);`

Shridharfly, 2017-03-15:
Thanks for the reply @BigMikes !
Instead of [this](https://github.com/contiki-os/contiki/blob/94854177e2837a6dec0e19624887165aa7cec7d5/examples/ipv6/rpl-udp/udp-client.c#L111) I tried  

`uip_udp_packet_sendto(client_conn, buf, MAX_PAYLOAD_LEN, &server_ipaddr, UIP_HTONS(UDP_SERVER_PORT));` 

But no success.

greg-king5, 2017-03-19:
You don't need to define `NETSTACK_CONF_WITH_IPV6`; the makefiles define it.

You might need to increase the size of the uIP buffer, so that it can hold your bigger payload.  Define a big-enough `UIP_CONF_BUFFER_SIZE` in the "project-conf.h" file (it needs to hold at least the payload and a header).

It looks as though you don't need much to trigger fragmentation.  Try payloads that are multiples of 128.

Shridharfly, 2017-03-20:
I was increasing size of payload but I partially fed data into buffer. As soon as I fed a lot of data in complete buf, the issue was solved.

closed 2017-03-20 by Shridharfly