contiki-os/contiki #2143: Multicast example cannot select a different multicast engine.

g-oikonomou, 2017-03-17:
The multicast example should be capable of switching between multicast engines merely by changing `UIP_MCAST6_CONF_ENGINE`. The multicast code under core should be able to automatically set RPL to operate in MOP 3 when required. This used to work but not any more. It appears to be related to the order of header file inclusion.

As a result of that, attempting to change the default multicast engine will lead to this error:

```
  CC        ../../../core/net/ipv6/uip6.c
In file included from ../../../core/net/ipv6/uip6.c:85:0:
../../../core/net/rpl/rpl-private.h:258:2: error: #error "RPL Multicast requires RPL_MOP_DEFAULT==3. Check contiki-conf.h"
 #error "RPL Multicast requires RPL_MOP_DEFAULT==3. Check contiki-conf.h"
  ^
make: *** [obj_srf06-cc26xx/uip6.o] Error 1
```

This was discovered during the discussion under #2101.

MohamedSeliem, 2017-03-17:
I have a question regarding this issue:

In order to set RPL_MOP_DEFAULT to 3 by including rpl-private.h :
  RPL_CONF_MOP must be 0 (undefined) and RPL_WITH_MULTICAST must be set to 1 (UIP_MCAST6_ENGINE = UIP_MCAST6_ENGINE_SMRF, or UIP_MCAST6_ENGINE = UIP_MCAST6_ENGINE_ESMRF)

However, In contiki-default-conf.h 
#ifndef RPL_CONF_MOP
#define RPL_CONF_MOP RPL_MOP_STORING_NO_MULTICAST
#endif /* RPL_CONF_MOP */

so from what i understand, by default if RPL_CONF_MOP is not defined in the project-conf for the example. it will be defined by contiki-default-conf.h 

which contradicts with the part that setting  RPL_MOP_DEFAULT to 3 in  rpl-private.h:
/* RPL Mode of operation */
#ifdef  RPL_CONF_MOP
#define RPL_MOP_DEFAULT                 RPL_CONF_MOP
#else /* RPL_CONF_MOP */
#if RPL_WITH_MULTICAST
#define RPL_MOP_DEFAULT                 RPL_MOP_STORING_MULTICAST
#else
#define RPL_MOP_DEFAULT                 RPL_MOP_STORING_NO_MULTICAST
#endif /* RPL_WITH_MULTICAST */
#endif /* RPL_CONF_MOP */

if i am not missing anything in here, then we should define RPL_CONF_MOP in uip-mcast6.h such that:
#if RPL_WITH_MULTICAST
#define RPL_MOP_DEFAULT                 RPL_MOP_STORING_MULTICAST
#endif

or we can simply erase the part in contiki-default-conf.h 
#ifndef RPL_CONF_MOP
#define RPL_CONF_MOP RPL_MOP_STORING_NO_MULTICAST
#endif /* RPL_CONF_MOP */

and let rpl-private.h defne the mode of operation.

???????

g-oikonomou, 2017-03-17:
I think you may have a point, but please include your code snippets inside code blocks: I am finding it somewhat difficult to understand what you are trying to say here :)

PR #2145 (2017-03-23): Don't force RPL MOP configuration in contiki-default-conf.h
We are trying to set the default MOP in two places:

* `rpl-private.h`: Considers multicast engine selection and correctly allows MOP to be set to 3 when required.
* `contiki-default-conf.h` is not only redundant but also fails to handle situations where RPL multicast is required.

When selecting a multicast engine that requires RPL to be in MOP 3, compilation fails with the error shown in #2143.

This pull simply removes RPL MOP-related configuration from `contiki-default-conf.h` and allows user-provided configuration to take care of [... cut]
closed 2017-03-23 by nvt