contiki-os/contiki #2152: Cannot compile rpl-tsch even in the InstantContiki VM

BowenWang, 2017-03-22:
Hi, 
I tried to compile the rpl-tsch example in contiki/examples/ipv6/rpl-tsch on my laptop, it didn't work, the error messages are:
make TARGET=sky
  CC        node.c
  CC        ../../../platform/sky/./contiki-sky-main.c
  LD        node.sky
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: node.sky section `.text' will not fit in region `rom'
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: node.sky section `.bss' will not fit in region `ram'
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: section .vectors loaded at [000000000000ffe0,000000000000ffff] overlaps section .text loaded at [0000000000004000,00000000000128f3]
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: region `ram' overflowed by 148 bytes
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: region `rom' overflowed by 15000 bytes
/usr/lib/gcc/msp430/4.6.3/mmpy-16/libcrt0.a(_copy_data.o): In function `__do_copy_data':
/build/buildd/gcc-msp430-4.6.3~mspgcc-20120406/./gcc-4.6.3/gcc/config/msp430/crt0.S:195: relocation truncated to fit: R_MSP430_16_BYTE against symbol `__data_load_start' defined in *ABS* section in node.sky
obj_sky/contiki-sky-main.o: In function `main':
/home/bowen/Bowen/gitrepo/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:140: relocation truncated to fit: R_MSP430_16 against symbol `memset' defined in .text section in /usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mmpy-16/libc.a(memset.o)
/home/bowen/Bowen/gitrepo/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:142: relocation truncated to fit: R_MSP430_16 against symbol `memcpy' defined in .text section in /usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mmpy-16/libc.a(memcpy.o)
/home/bowen/Bowen/gitrepo/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:154: relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
/home/bowen/Bowen/gitrepo/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:154: relocation truncated to fit: R_MSP430_16 against symbol `printf' defined in .text section in /usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mmpy-16/libc.a(printf.o)
/home/bowen/Bowen/gitrepo/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:156: relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
/home/bowen/Bowen/gitrepo/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:156: relocation truncated to fit: R_MSP430_16 against symbol `printf' defined in .text section in /usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mmpy-16/libc.a(printf.o)
/home/bowen/Bowen/gitrepo/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:158: relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
/home/bowen/Bowen/gitrepo/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:158: relocation truncated to fit: R_MSP430_16 against symbol `printf' defined in .text section in /usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mmpy-16/libc.a(printf.o)
/home/bowen/Bowen/gitrepo/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:274: relocation truncated to fit: R_MSP430_16 against symbol `memset' defined in .text section in /usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mmpy-16/libc.a(memset.o)
/home/bowen/Bowen/gitrepo/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:276: additional relocation overflows omitted from the output
collect2: ld returned 1 exit status
../../../Makefile.include:280: recipe for target 'node.sky' failed
make: *** [node.sky] Error 1
rm node.co obj_sky/contiki-sky-main.o

at first, I think it may be a compiler error because that my msp430-gcc is 4.6.3 version. Then I switch to the InstantContiki VM provided on the official website, the error messages are slightly different:
make TARGET=sky
mkdir obj_sky
  CC        ../../../apps/orchestra/orchestra.c
  CC        ../../../apps/orchestra/orchestra-rule-default-common.c
  CC        ../../../apps/orchestra/orchestra-rule-eb-per-time-source.c
  CC        ../../../apps/orchestra/orchestra-rule-unicast-per-neighbor-rpl-storing.c
  CC        ../../../apps/orchestra/orchestra-rule-unicast-per-neighbor-rpl-ns.c
  CC        ../../../platform/sky/./contiki-sky-platform.c
  CC        ../../../dev/sht11/sht11.c
  CC        ../../../dev/sht11/sht11-sensor.c
  CC        ../../../platform/sky/dev/light-sensor.c
  CC        ../../../platform/sky/dev/battery-sensor.c
  CC        ../../../platform/sky/dev/button-sensor.c
  CC        ../../../platform/sky/dev/radio-sensor.c
  CC        ../../../cpu/msp430/f1xxx/spi.c
  CC        ../../../dev/ds2411/ds2411.c
  CC        ../../../platform/sky/dev/xmem.c
  CC        ../../../platform/sky/dev/i2c.c
  CC        ../../../platform/sky/./node-id.c
  CC        ../../../core/lib/sensors.c
  CC        ../../../core/cfs/cfs-coffee.c
  CC        ../../../dev/cc2420/cc2420.c
  CC        ../../../cpu/msp430/./cc2420-arch.c
  CC        ../../../cpu/msp430/./cc2420-arch-sfd.c
  CC        ../../../platform/sky/dev/sky-sensors.c
  CC        ../../../cpu/msp430/./uip-ipchksum.c
  CC        ../../../cpu/msp430/f1xxx/uart1.c
  CC        ../../../cpu/msp430/./slip_uart1.c
  CC        ../../../cpu/msp430/dev/uart1-putchar.c
  CC        ../../../core/lib/me.c
  CC        ../../../core/lib/me_tabs.c
  CC        ../../../core/dev/slip.c
  CC        ../../../core/lib/crc16.c
  CC        ../../../cpu/msp430/f1xxx/msp430.c
  CC        ../../../cpu/msp430/./flash.c
  CC        ../../../cpu/msp430/f1xxx/clock.c
  CC        ../../../core/dev/leds.c
  CC        ../../../cpu/msp430/./leds-arch.c
  CC        ../../../cpu/msp430/./watchdog.c
  CC        ../../../cpu/msp430/./lpm.c
[... 110 lines cut ...]
/usr/local/msp430/bin/../lib/gcc/msp430/4.7.0/../../../../msp430/bin/ld: node.sky section `.text' will not fit in region `rom'
/usr/local/msp430/bin/../lib/gcc/msp430/4.7.0/../../../../msp430/bin/ld: node.sky section `.bss' will not fit in region `ram'
/usr/local/msp430/bin/../lib/gcc/msp430/4.7.0/../../../../msp430/bin/ld: section .vectors loaded at [0000ffe0,0000ffff] overlaps section .text loaded at [00004000,00012565]
/usr/local/msp430/bin/../lib/gcc/msp430/4.7.0/../../../../msp430/bin/ld: region `ram' overflowed by 142 bytes
/usr/local/msp430/bin/../lib/gcc/msp430/4.7.0/../../../../msp430/bin/ld: region `rom' overflowed by 14098 bytes
/usr/local/msp430/bin/../lib/gcc/msp430/4.7.0/mmpy-16/libcrt0.a(_copy_data.o): In function `__do_copy_data':
/home/user/building-mspgcc/gcc-4.7.0-msp430/msp430/mmpy-16/libgcc/../../../../gcc-4.7.0/libgcc/config/msp430/crt0.S:208:(.init4+0x12): relocation truncated to fit: R_MSP430_16_BYTE against symbol `__data_load_start' defined in *ABS* section in node.sky
obj_sky/contiki-sky-main.o: In function `main':
/home/user/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:140:(.init9+0x84): relocation truncated to fit: R_MSP430_16 against symbol `memset' defined in .text.memset section in /usr/local/msp430/bin/../lib/gcc/msp430/4.7.0/../../../../msp430/lib/mmpy-16/libc.a(memset.o)
/home/user/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:142:(.init9+0x90): relocation truncated to fit: R_MSP430_16 against symbol `memcpy' defined in .text.memcpy section in /usr/local/msp430/bin/../lib/gcc/msp430/4.7.0/../../../../msp430/lib/mmpy-16/libc.a(memcpy.o)
/home/user/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:153:(.init9+0x96): relocation truncated to fit: R_MSP430_16 against symbol `linkaddr_set_node_addr' defined in .text.linkaddr_set_node_addr section in contiki-sky.a(linkaddr.o)
/home/user/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:154:(.init9+0x9a): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
/home/user/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:154:(.init9+0x9e): relocation truncated to fit: R_MSP430_16 against symbol `printf' defined in .text.printf section in /usr/local/msp430/bin/../lib/gcc/msp430/4.7.0/../../../../msp430/lib/mmpy-16/libc.a(printf.o)
/home/user/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:156:(.init9+0xae): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
[... cut ...] al/msp430/bin/../lib/gcc/msp430/4.7.0/../../../../msp430/lib/mmpy-16/libc.a(printf.o)
/home/user/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:274:(.init9+0xe6): additional relocation overflows omitted from the output
collect2: error: ld returned 1 exit status
make: *** [node.sky] Error 1
rm node.co obj_sky/contiki-sky-main.o

Is that because something wrong with the compiler, project configuration or it is just the target sky cannot provide enough ram and rom?

Bowen

g-oikonomou, 2017-03-23:
Can you try building for Z1?

BowenWang, 2017-04-04:
I compile this using 
make TARGET=z1
it is still not working, the error information is almost the same as the sky target
section .data will not fit in region 'rom'
region rom overflowed by 2 bytes

Is that possible to run this application on Telosb?

ngovanmao, 2017-06-15:
You can compile well with openmote-cc2538 or zoul platforms which have larger ROM than SKY or Z1.

lamnd09, 2017-06-20:
Hi, your problem is that you are using the old version of MSP430. Try with MSP430 4.7.2 or higher.