contiki-os/contiki #2163: Fail to execute tools/motelist-linux in Sensor Data Collect

aitsgehu, 2017-03-31:
Hello,
 In  cooja, when i am using the tool COLLECT VIEW  there is a error. I cant run Program Nodes in Sensor Data Collect with Contiki  and i have error that show me cant find file ./tools/motelist-linux

 I search put i cant find anything

thanks

greg-king5, 2017-03-31:
> In cooja, when I am using the tool `COLLECT VIEW`, there is an error. I can't run `Program Nodes` in `Sensor Data Collect` with Contiki; and, I have an error that shows me, "can't find file `./tools/motelist-linux`".

It's in the `sky/` subdirectory.  So, change the paths in
`"tools/collect-view/src/org/contikios/contiki/collect/MoteFinder.java"` (e.g.,
`"./tools/sky/motelist-linux"`).

davidcawork, 2018-11-09:
> It's in the `sky/` subdirectory. So, change the paths in
> `"tools/collect-view/src/org/contikios/contiki/collect/MoteFinder.java"` (e.g.,
> `"./tools/sky/motelist-linux"`).

I did it, and I still have the same error :/  Someone Have fix it ?

Moneymonkey95, 2019-11-06:
Thanks you @greg-king5.
It worked for me.

kipongo, 2022-07-15:
> 

@Moneymonkey95 how did you solve the issue "java.io.IOException: Failed to execute './tools/motelist-linux'" ??
Please help @greg-king5