contiki-os/contiki #2169: connect to MQTT broker locally

B00067132, 2017-04-02:
I followed this tutorial :
[http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples](url)

but instead of using the cc26xx-demo I used the cc26xx-web-demo and successfully manged to get everything up and running and I can access the 6lbr web page, when I access the sensorTag page I see a mqtt configuration page as shown: 
[image]

I'm trying to locally connect to the mqtt broker using javascript and have it send the sensor data so I can print it to the console.

```
my js file 
var mqtt = require('mqtt');
var fs = require('fs');

var PORT = 1883; 
var HOST = '0064:ff9b:0000:0000:0000:0000:b8ac:7cbd';

var options = { port: PORT, host: HOST, };
var client = mqtt.connect(options);

client.on('connect', function ()
{
        console.log("it arrived at 1")
    client.subscribe("topic, command or data");
        console.log("subbed")

    client.publish("topic, command or data", data, function () {
        console.log("published")
    });

});

client.on('error', function () {console.log("error") });

client.on('offline', function () { console.log("off-line")});

client.on('message', function (topic, message) {

    console.log(message.toString());
});
```

but this is the output I'm getting 
[image]

I ran tcpdump on my edge router (beaglebone black wirless) to listen for traffic on port 1883 and it seems to be picking up packets as shown 
[image]

the only things I changed in the web demo before I complied it is  in the project-conf.h
```
/* Change to match your configuration */
#define IEEE802154_CONF_PANID            0xABCD
#define RF_CORE_CONF_CHANNEL                 26
#define RF_BLE_CONF_ENABLED                   0

```
and then flashed it to my sensorTag cc2650stk

is the mqtt broker Ip  and port I'm connecting to, right? 

my setup 

BBB wirless : edge router running 6lbr 
using the CC2531EMK dongle flashed with  cc2531-slip-radio_contikimac from the ti example above
cc2650STK : flashed with cc26xx-web-demo with mention change I made to the project-conf.h

6lbr settings :

```
MODE=ROUTER

RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=wlan0
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0 
BAUDRATE=115200
LOG_LEVEL=3
```

jonnteolsson, 2017-04-07:
You have a reply on your post on the TI E2E forum.

kaleemullah360, 2017-04-09:
Have you installed Mosquitto Broker, and make sure that Mosquitto broker service is running. the MQTT library uses it. be sure to start service like this.

`sudo service mosquitto start`

you can also check its status to check if it is running properly.
`sudo service mosquitto status`
or extra options are
`Usage: /etc/init.d/mosquitto {start|stop|reload|force-reload|restart|try-restart|status}`

closed 2017-04-24 by g-oikonomou

rahultt007, 2017-07-06:
Hi,

We are also trying the same thing but with the BBB having Ethernet interface. We started the mosquitto and can see the log data publish and subcribe.

debian@beaglebone:~$ mosquitto
1499344364: mosquitto version 1.4.8 (build date 2017-07-04 12:05:37+0000) starting
1499344364: Using default config.
1499344364: Opening ipv4 listen socket on port 1883.
1499344364: Opening ipv6 listen socket on port 1883.
1499344393: New connection from ::1 on port 1883.
1499344393: New client connected from ::1 as mosqsub/1328-beaglebone (c1, k60).
1499344421: New connection from ::1 on port 1883.
1499344421: New client connected from ::1 as mosqpub/1342-beaglebone (c1, k60).
1499344421: Client mosqpub/1342-beaglebone disconnected.

6lbr configuration is as follows,

MODE=ROUTER
#MODE=SMART-BRIDGE
#MODE=RPL-RELAY
#MODE=FULL-TRANSPARENT-BRIDGE
#MODE=NDP-ROUTER
#MODE=6LR
#MODE=RPL-ROOT

RAW_ETH=1
BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=eth0
#RAW_ETH_FCS=1
RAW_ETH_FCS=0

DEV_RADIO=/dev/ttyACM0 #2.4GHz USB Dongle will enumerate like ACM0
#DEV_RADIO=/dev/ttyUSB1 #SmartRF06EB+CC13xxEM will enumerate like USB1
BAUDRATE=115200

LOG_LEVEL=3

and interface configuration is as follows,

# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).

# The loopback network interface
auto lo
iface lo inet loopback

# The primary network interface
#auto br0
#iface br0 inet dhcp
auto eth0
iface eth0 inet dhcp

# Example to keep MAC address between reboots
#hwaddress ether DE:AD:BE:EF:CA:FE

# The secondary network interface
#auto eth1
#iface eth1 inet dhcp

# WiFi Example
#auto wlan0
#iface wlan0 inet dhcp
#    wpa-ssid "essid"
#    wpa-psk  "password"
# Ethernet/RNDIS gadget (g_ether)
# ... or on host side, usbnet and random hwaddr
# Note on some boards, usb0 is automaticly setup with an init script
iface usb0 inet static
    address 192.168.7.2
    netmask 255.255.255.0
    network 192.168.7.0
    gateway 192.168.7.1

Please let me know the issue.

Thanks
Rahul