contiki-os/contiki #2177: Outdoor mesh range with the CC1310

jbartlett12026, 2017-04-06:
I am running the 6lbr border router demo using CC1310s and following the http://processors.wiki.ti.com/index.php/Cc26xx_sw_examples and have stable running connections indoors. However, when I go outside to test, the connections are hardly established at a meter distance. I am using duck antennas on the nodes and slip radio and I wonder if it is a factor of the output power. I noticed that there have been issues with the CC1310 RF driver, but is there any way to better control transmitting output power?

g-oikonomou, 2017-04-12:
Try with #1931 and #1932 (I'll be merging them in a few days).

g-oikonomou, 2017-04-24:
@jbartlett12026 what's the status here. Is this working better for you or shall we keep this open?

jbartlett12026, 2017-04-24:
@g-oikonomou Yes it is! For those using CC1310 LPs: There were three things that seemed to increase range/connectivity. Using null_rdc over contiki_mac, using the #1932 pull over the main, and changing the VDDH register to broadcast at 14dbm.

g-oikonomou, 2017-04-24:
Fantastic. Closing

g-oikonomou, 2017-04-24:
I am still unhappy with the fact that nullrdc increases range - this shouldn't be happening. I think the radio behaves differently shortly after power up compared to when it's been running for a while. We need to figure this out...

closed 2017-04-24 by g-oikonomou