contiki-os/contiki #2179: Problem reading input ADC CC26XX

aignacio, 2017-04-07:
Hello all,

Did anybody try to read another ADC Input different from **IO7 (DIO23)** in the **CC2650** target? I try to read the others inputs but seems not working, the ADC sample always the same voltage and not sense on different stimulous... I try to read this inputs:
```
ADC_COMPB_IN_AUXIO6
ADC_COMPB_IN_AUXIO5
ADC_COMPB_IN_AUXIO4
ADC_COMPB_IN_AUXIO3
ADC_COMPB_IN_AUXIO2
ADC_COMPB_IN_AUXIO1
ADC_COMPB_IN_AUXIO0
```

The code that used to read is below:

```
uint16_t readADCFieldSensor(void){
  uint16_t singleSample;
  ti_lib_aon_wuc_aux_wakeup_event(AONWUC_AUX_WAKEUP);
  while(!(ti_lib_aon_wuc_power_status_get() & AONWUC_AUX_POWER_ON));
  ti_lib_aux_wuc_clock_enable(AUX_WUC_ADI_CLOCK | AUX_WUC_ANAIF_CLOCK | AUX_WUC_SMPH_CLOCK);
  while(ti_lib_aux_wuc_clock_status(AUX_WUC_ADI_CLOCK | AUX_WUC_ANAIF_CLOCK | AUX_WUC_SMPH_CLOCK) != AUX_WUC_CLOCK_READY);

  AUXADCSelectInput(ADC_COMPB_IN_AUXIO5); // <--- Change the ADC inputs here
  AUXADCEnableSync(AUXADC_REF_FIXED,  AUXADC_SAMPLE_TIME_2P7_US,   AUXADC_TRIGGER_MANUAL);
  AUXADCGenManualTrigger();
  singleSample = AUXADCReadFifo();
  AUXADCDisable();
  return singleSample;
}

The only input that works seems to be the IO7...
```

g-oikonomou, 2017-04-08:
I can't see anything obviously wrong with your code. Have you tried using the ADC sensor driver available in the CPU dir? We have not had any such problems here.

Is any other part of your code messing with the pins? What board are you doing this with?

aignacio, 2017-04-08:
I didn't try to use the ADC sensor driver and I'll try now, I'm using CC2650 launchpad and none part of code mess with GPIO's....

aignacio, 2017-04-08:
I tried to use the ADC sensor driver but it seems to show the same thing before, I also tried to use another CC2650 Launchpad but result in the same value...not change on stimulous. 
[image]

chenek, 2017-04-10:
Latest cc26xx-web-demo example includes DIO23 as ADC reading. You can git pull latest code and you can enable it by define "CC26XX_WEB_DEMO_CONF_ADC_DEMO" to 1 in project-conf.h

aignacio, 2017-04-10:
@chenek thanks, but the DIO23 is the only ADC input that I can read ok, not with ADC Sensor driver but with the first function setting as:

```
uint16_t readADCFieldSensor(void){
  uint16_t singleSample;
  ti_lib_aon_wuc_aux_wakeup_event(AONWUC_AUX_WAKEUP);
  while(!(ti_lib_aon_wuc_power_status_get() & AONWUC_AUX_POWER_ON));
  ti_lib_aux_wuc_clock_enable(AUX_WUC_ADI_CLOCK | AUX_WUC_ANAIF_CLOCK | AUX_WUC_SMPH_CLOCK);
  while(ti_lib_aux_wuc_clock_status(AUX_WUC_ADI_CLOCK | AUX_WUC_ANAIF_CLOCK | AUX_WUC_SMPH_CLOCK) != AUX_WUC_CLOCK_READY);

  AUXADCSelectInput(ADC_COMPB_IN_AUXIO7); // <-- This input reads ok
  AUXADCEnableSync(AUXADC_REF_FIXED,  AUXADC_SAMPLE_TIME_2P7_US,   AUXADC_TRIGGER_MANUAL);
  AUXADCGenManualTrigger();
  singleSample = AUXADCReadFifo();
  AUXADCDisable();
  return singleSample;
}
```

aignacio, 2017-04-10:
Code used to test attach....
[leaf.zip](https://github.com/contiki-os/contiki/files/908935/leaf.zip)

chenek, 2017-04-10:
Just use hello-world to test ADC_COMPB_IN_AUXIO6 which is connected to DIO24 on LAUNCHXL-CC2650 and it works fine. You can refer to my hello-world .c

[hello-world.zip](https://github.com/contiki-os/contiki/files/909003/hello-world.zip)

aignacio, 2017-04-10:
This examples seems work nice but I notice another problem with the ADC read, I observed a low impedance on ADC input's that drops my NTC thermistor voltage, can you confirm the diagnoses? @chenek and @g-oikonomou

[image]

g-oikonomou, 2017-04-24:
Hi @aignacio I think by reading between the lines you are trying to use the code for one of Contiki's boards to power a custom board? If you are using a pin for ADC, you should definitely not include it in the list of pins that get configured by `configure_unused_pins()`, since this function configures pins to a low energy-leakage state.

You are right that the ADC driver doesn't disable pull, but it can be disabled as you suggest: `ti_lib_ioc_io_port_pull_set(..., IOC_NO_IOPULL);`.

I cannot comment on your design. Please close this when you no longer have problems.

aignacio, 2017-04-27:
Hi @g-oikonomou , you're right I'm trying to read ADC input in a custom CC2650 board with CC2650 5X5  IC and now after doing the mods all works right, thanks for support...

closed 2017-04-27 by aignacio