contiki-os/contiki #2181: What's the difference between Contiki and Contiki3.0?

Windywind, 2017-04-08:
As the title said, I am confused by these.

I am a MSc Student trying to do some research on CoAP using Contiki, and I came across with this article today:
[http://anrg.usc.edu/contiki/index.php/REST_example_running_on_Cooja_and_Sky_motes]

It uses the files under  /contiki-2.7/examples/rest-example folder, but I cannot find this "rest-example" folder in my InstantContiki3.0, but just some "er-rest-example". Then I realized the auther of the page is using Contiki2.7.

So, any suggestions on what part of codes should I read if I was tuning CoAP? :D 

For example, if I'd like to change the value of `ACK_TIMEOUT`, `ACK_RANDOM_FACTOR`.

Thanks in advance.

JoaoS, 2017-04-12:
Hi, i also found that website some time ago and think some things might be outdated you can try asking your question in the mailing list you will probably get better luck.
contiki-developers@lists.sourceforge.net

Windywind, 2017-04-12:
Hi # @JoaoS 

How to use this mailing list? Just directly fire out the question? LOL~ I have never tried this before. Please tell me…

JoaoS, 2017-04-12:
first you register you email address here https://lists.sourceforge.net/lists/listinfo/contiki-developers
then from that email account you send an email to that address with your subject.
Then daily (I think) you will receive all messages from developers who sent questions to the list, you can send questions and reply to those questions

closed 2017-04-12 by g-oikonomou

Windywind, 2017-04-12:
Hey @g-oikonomou , still nobody answered the question