contiki-os/contiki #2184: make login problem

domsca, 2017-04-10:
Hi to everybody,

I'm a new user of this community and in programming, so sorry if I'm make sorry and please try to be as clear as possible, for me everything is new.
I have problem to complete the first example to start contiki with zolertia z1.
In particular I have problem with `make login`.
This is the output I receive (I'm working with Windows and IstantContiki3.0).
```
../../tools/sky/serialdump-linux -b115200 /dev/ttyUSB0
connecting to /dev/ttyUSB0 (115200)
open: Invalid argument
make: *** [login] Error 255

```
I tried a lot of solutions alredy proposed, but I still have this problem. I tried with : `apt-get install libc6:i386` and `apt-get install ia32-libs` but this is the output I receive `Unable to locate package ia32-libs`.

Can someone help me?

lrjoyce, 2017-06-02:
(I'm new to the community so take what I say with a grain of salt.)

I'm in the same boat. I also had the same problem with `ia32-libs`. I think `ia32-libs` is outdated. Instead, try `lib32ncurses5` and `lib32z1`. 

However, I'm still receiving the same error you are, so I don't think this is the fix (although I think it fixed another error for me). I'm trying the `hello-world` program on a Tmote sky (running ubuntu 16.04 and InstantContiki3.0), I get the output:
`using saved target 'sky'
../../tools/sky/serialdump-linux -b115200 /dev/ttyUSB0
connecting to /dev/ttyUSB0 (115200)
open: Invalid argument
../../platform/sky/Makefile.common:193: recipe for target 'login' failed
make: *** [login] Error 255
`

greg-king5, 2017-06-03:
That program binary seems to be old.  I think that you must rebuild it:
```sh
cd tools/sky
rm serialdump-linux; make
```

elal007, 2017-06-15:
Hi All,
I also have same problem. Could you please tell me, where  i can get a new one. I got the serialdump-linux from https://github.com/contiki-os/contiki/blob/master/tools/sky/serialdump-linux.

greg-king5, 2017-06-15:
You must create the new one.  Use the commands that I gave in my previous message.

orgosoft, 2017-09-23:
Dear folks
I had the same problem I followed the above instructions and have apparently solved that issue, but now mine appears to have made login but nothing appears on console... underneath is the working
ali@3li-machine:~/contiki/examples/hello-world$ sudo make hello-world.upload && make z1-reset && make login
using saved target 'z1'
fatal: Not a git repository: '../../.git'
msp430-objcopy hello-world.z1 -O ihex hello-world.ihex
cp hello-world.ihex tmpimage.ihex
/dev/ttyUSB0
make z1-reset z1-upload
make[1]: Entering directory '/home/ali/contiki/examples/hello-world'
using saved target 'z1'
fatal: Not a git repository: '../../.git'
make -k -j 20 z1-reset-sequence
make[2]: Entering directory '/home/ali/contiki/examples/hello-world'
using saved target 'z1'
fatal: Not a git repository: '../../.git'
../../tools/z1/z1-bsl-nopic --z1 -c /dev/ttyUSB0 -r
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Use -h for help
Use --fromweb to upgrade a GoodFET.
Reset device ...
Done
make[2]: Leaving directory '/home/ali/contiki/examples/hello-world'
make -j 20 z1-upload-sequence
make[2]: Entering directory '/home/ali/contiki/examples/hello-world'
using saved target 'z1'
fatal: Not a git repository: '../../.git'
+++++ Erasing /dev/ttyUSB0
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Use -h for help
Use --fromweb to upgrade a GoodFET.
Mass Erase...
Transmit default password ...
+++++ Programming /dev/ttyUSB0
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Changing baudrate to 38400 ...
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Changing baudrate to 38400 ...
Traceback (most recent call last):
  File "../../tools/z1/z1-bsl-nopic", line 1940, in <module>
    main(0);
  File "../../tools/z1/z1-bsl-nopic", line 1865, in main
    speed=speed,
  File "../../tools/z1/z1-bsl-nopic", line 1241, in actionStartBSL
    self.actionChangeBaudrate(speed)            #change baudrate
  File "../../tools/z1/z1-bsl-nopic", line 1368, in actionChangeBaudrate
    self.serialport.setBaudrate(baudrate)
AttributeError: 'Serial' object has no attribute 'setBaudrate'
+++++ Resetting /dev/ttyUSB0
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Use -h for help
Use --fromweb to upgrade a GoodFET.
Reset device ...
Done
make[2]: Leaving directory '/home/ali/contiki/examples/hello-world'
make[1]: Leaving directory '/home/ali/contiki/examples/hello-world'
rm hello-world.ihex
using saved target 'z1'
fatal: Not a git repository: '../../.git'
make -k -j 20 z1-reset-sequence
make[1]: Entering directory '/home/ali/contiki/examples/hello-world'
using saved target 'z1'
fatal: Not a git repository: '../../.git'
../../tools/z1/z1-bsl-nopic --z1 -c /dev/ttyUSB0 -r
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Use -h for help
Use --fromweb to upgrade a GoodFET.
Reset device ...
Done
make[1]: Leaving directory '/home/ali/contiki/examples/hello-world'
using saved target 'z1'
fatal: Not a git repository: '../../.git'
../../tools/sky/serialdump-linux -b115200 /dev/ttyUSB0
connecting to /dev/ttyUSB0 (115200) [OK]

i shall be grateful...

shaik2a, 2017-10-23:
@orgosoft  I am also getting the same issue. 
Did you fix the issue? If so kindly do let me know. 
Thank you.

orgosoft, 2017-10-23:
There is a solution on internet. Plz google. It was about updating an old
function in python script
Good luck.

eesati, 2019-12-01:
instead of make login try this ../../tools/sky/serialdump-linux -b115200 /dev/ttyUSBx
x - ur com port. 

make login is may be stuck with /dev/ttyUSB0 which is not existing, it may be at /dev/ttyUSB1 for example.

sarmasid, 2022-05-17:
> There is a solution on internet. Plz google. It was about updating an old function in python script Good luck.

Can you kindly tell me how you solved this problem? I already checked with different serial terminal tools (cutecom, minicom, coolterm etc.). But, nothing appeared on the serial terminal!