contiki-os/contiki #2185: Possibly unterminated strings in resolv.c

tonnenpinguin, 2017-04-11:
My static code analysis tool shows several possibly unterminated string warnings for resolv.c, due to the use of [strncpy ](http://www.cplusplus.com/reference/cstring/strncpy/)- which does not implicitly add a null terminator if the original string is longer than len.

This issue occurs in lines [1092](https://github.com/contiki-os/contiki/blob/master/core/net/ip/resolv.c#L1092), [1250](https://github.com/contiki-os/contiki/blob/master/core/net/ip/resolv.c#L1250), the latter of which causes several other functions to have possibly undefined  behaviour.

A relatively easy fix would be to switch to BSD's [strlcpy](http://man.openbsd.org/strlcpy), however, I'm not sure if this is available for all platforms.

greg-king5, 2017-04-11:
Strictly speaking, it isn't a bug because those buffers are one character bigger than the defined limit.  And, they are stored statically.  Therefore, the last byte in those buffers already is `'\0'`.  The `strncpy()` calls don't change it.

The actual bug is in some of the `strncat()` calls.  They assume that the length parameter describes the target buffer -- but, it actually is the maximum number of characters to copy; it doesn't count what already is in the target buffer.  Therefore, those calls can overflow those targets!

tonnenpinguin, 2017-04-13:
You're right. I've changed my pr to handle the cases you described!

PR #2189 (2017-04-18): Fix potentially unterminated strings
This fixes some potential issues with strings being copied using strncpy, which causes char arrays that are used as strings later on to be non-terminated.

I have also changed the typecasts in the tolower function calls to unsigned char.

Fixes #2185
closed 2017-04-18 by nvt