contiki-os/contiki #2191: Contiki IPv4 connection to 127.0.0.1 loop back address

kajaldesai123, 2017-04-12:
Hello sir,
I m new to contiki. I used cc2650 sensortag and now i m using cc1310 launchpad. i use beagebone as a edge router and one cc1310 launchpad as a slipradio. everything works fine. i can open Coap and web and also i can get sensors data on web and coap.

now, i want to send sensor data on our own ipv4 cloud using tcp socket. but, i cant create tcp socket to my cloud. Is it possible to create ipv4 tcp socket directly from node?
can anyone suggest me how to create ipv4 tcp socket to 127.0.0.1 ip address and 1502 port?

g-oikonomou, 2017-04-12:
https://github.com/contiki-os/contiki/wiki#Help

closed 2017-04-12 by g-oikonomou

kajaldesai123, 2017-04-14:
can you please reopen the issue? I could not make tcp socket in ipv4 address. I already try to connect from last 6 month using http-socket and p-sock library but, i couldn't. @g-oikonomou

greg-king5, 2017-04-15:
The Contiki community wants people to ask questions like yours on the mailing list instead of the issue tracker.  (This tracker is for bug reports and discussions about code improvement.)