contiki-os/contiki #2192: Sniffer "Decrypt failed" with LLSEC

Alexor86, 2017-04-12:
Hello!

I have several nodes, 6lbr and sniffer (from https://github.com/g-oikonomou/sensniff). Everything works well, but if I enable LLSEC - the sniffer can't decrypt packets. Nodes and 6lbr with LLSEC work well.

Node config (from project-conf.h):

```
#undef LLSEC802154_CONF_ENABLED
#define LLSEC802154_CONF_ENABLED          1
#undef NETSTACK_CONF_FRAMER
#define NETSTACK_CONF_FRAMER              noncoresec_framer
#undef NETSTACK_CONF_LLSEC
#define NETSTACK_CONF_LLSEC               noncoresec_driver
#undef NONCORESEC_CONF_SEC_LVL
#define NONCORESEC_CONF_SEC_LVL           6
#define NONCORESEC_CONF_KEY { 0x00 , 0x01 , 0x02 , 0x03 , \
                              0x04 , 0x05 , 0x06 , 0x07 , \
                              0x08 , 0x09 , 0x0A , 0x0B , \
                              0x0C , 0x0D , 0x0E , 0x0F }
```

6lbr config (from nvm.dat):

```
Security layer : 1
Security level : 6
Security key : 0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f
```

Wireshark config:
[image]

Wireshark result:
[image]

g-oikonomou, 2017-04-12:
Can't say I've ever tried that! Are you sure you are setting the key correctly in WS? Are those `:` characters required between bytes?

Alexor86, 2017-04-12:
I get the same result with "000102030405060708090A0B0C0D0E0F" and "000102030405060708090a0b0c0d0e0f" keys.

MartinNPN, 2017-05-30:
Hello,

i didn't test it, but have you tried to convert the Key from hex values to characters ?
As i see the key is stored as char array at wireshark, so maybe it is only an conversion problem.