contiki-os/contiki #2198: Failed to execute './tools/motelist-linux'

Shridharfly, 2017-04-17:
I am using sky motes in cooja simulator of contiki. I want to use `collect-view`. So I added few sky motes in a simulation and right clicked one of the nodes to start `collect-view`. Then I clicked 'Program-Nodes' button. I got the following error:

    Programming failed: java.io.IOException: Failed to execute './tools/motelist-linux'.

For sky motes, I noticed the `motelist-linux` file is [here][1]. So I updated the [lines][2] to 

    public static final String MOTELIST_LINUX = "./tools/sky/motelist-linux";
    public static final String MOTELIST_MACOS = "./tools/sky/motelist-macos";

I have verified that `motelist-linux` & `motelist-macos` files have necessary permissions. But I got the same error again.

    Programming failed: java.io.IOException: Failed to execute './tools/sky/motelist-linux'.

How do I get rid of the error? or Is there any other way to use `collect-view`?

  [1]: https://github.com/contiki-os/contiki/tree/master/tools/sky
  [2]: https://github.com/contiki-os/contiki/blob/master/tools/collect-view/src/org/contikios/contiki/collect/MoteFinder.java#L56-L57

vhkrishna, 2017-04-24:
You need sudo permissions for accessing serial port in linux. Please open cooja with sudo and try. it might work. There is another possibility that serial port of the mote might be opened by another application. Make sure that no other application is using serial port of the mote which you are trying to porgram.

closed 2017-05-07 by Shridharfly

Suelencremonini, 2017-11-09:
I opened Cooja using sudo, didn't work. I checked the ports being used and the mote's one wasn't there... Any other idea?

vhkrishna, 2017-11-10:
I think PATH variables might be and issue. some of the PATH variables which
are available as normal user might not be available when you are in super
user. Try opening cooja as normal user.

ghadaj1, 2018-01-27:
Hello 
I am using contiki too. I had the same problem. I tried to follow the advices but it still doesn't work. 
Did you find a solution @Shridharfly
Thank you

Nadeem7hah, 2020-02-13:
I am also using contiki and faceing same problem. Any have solution please help me.

hknaydin, 2020-02-13:
Hello, I want to look at the pc via teamviewer when available.
I solved a few people like this.

Nadeem7hah <notifications@github.com>, 13 Şub 2020 Per, 17:22 tarihinde
şunu yazdı: