contiki-os/contiki #2199: TCPIP stack crashing

saurabhbansal90, 2017-04-20:
I am using Contiki OS on CC2650 from TI. There I am using udp-server.c and rpl-border-router.c for communication between them.
For communication I am creating sockets and sending and recieving the data continously. But this communication is stopping after approx. 4000 transactions. I am creating a timer also and printing some data in serial.
The reason for communication stop seems to be TCPIP stack crash, as the microcontroller is running after communication stopping also.
So can somebody help me how to debug it further and resolve the issue.

saurabhbansal90, 2017-04-24:
Can anyone provide the information like how long the communication has been done with Contiki OS. I think it is issue with stack in Contiki OS as after communication stops, the micro is still working.

niziak, 2017-04-24:
You should compile some contiki .c files with debug enabled to see where communication stops. You have to modify .c files (e.g. /core/net/mac/sicslowmac/sicslowmac.c) to change lines with `#define DEBUG 0`

We are using Contiki with CC1310 for days. Hundreds of megabytes transferred without problem(after I've fixed the radio with #2161)

saurabhbansal90, 2017-04-24:
Hello Niziak, thanks a lot for guidance. I will look into the issue as suggested.

saurabhbansal90, 2017-04-26:
Hello Niziak, I am trying to resolve this issue and I am having below setup -
- Linux board, running TunSLIP6
- CC2650 running **rpl-border-router** and interfaced with linux board by SLIP protocol
- CC2650 running **udp-server** program

I am creating a tunnel in my Linux board, and pinging to **udp-server** node. I observed that the ping stops for sometime and start getting responses again, like below -
64 bytes from aaaa::212:4b00:79e:8f87: seq=**16** ttl=63 time=61.401 ms
64 bytes from aaaa::212:4b00:79e:8f87: seq=**46** ttl=63 time=158.386 ms

Please observe the seq id's. So for sometime I am not getting ping response on my Linux console, but actually it seems to be happeing.
I tried to debug the Border-Router code and found there is a function **static int output(void)** which is called in **core/net/ip/tcpip.c** file at line 578. This function is responsible to send recieved packet to serial by SLIP. I observed this function **static int output(void)** is not being called from **tcpip.c** for sometime and it start being called after sometime.

Can someone please tell me the possible reason for this behavior or some flags that I can check to debug it further.

niziak, 2017-04-26:
I don't know, try to add "define NETSTACK_CONF_RDC nullrdc_driver" in project-conf.h

saurabhbansal90, 2017-05-01:
Hello Niziak,
I did some testing with enabling logs in tcpip.c file. And there I got below output:

_tcpip_ipv6_output: no route found, using default route
tcpip_ipv6_output: Destination off-link but no route_

This seems to be a configuration issue, rather any bug in stack. So can you please give me some hint to solve this issue.

saurabhbansal90, 2017-05-03:
I am using the latest contiki source now, and it doesn't possess any such issue.
Thanks for your support, Niziak.

closed 2017-05-10 by saurabhbansal90