contiki-os/contiki #2201: Issue when I set MQTT interval to 480 in cc26xx-web-demo MQTT/IBM Cloud Config page

chenek, 2017-04-21:
When I set MQTT interval to 480 in cc26xx-web-demo MQTT/IBM Cloud Config page, I see the report duration is always about 200 seconds in IBM quickstart. I try to enable debug message in MQTT and find it would disconnect about 200 seconds with the following information in UART. I suspect this is about MQTT keep alive implementation. Does anyone see similar issue? 

MQTT - Disconnected by tcp event 3
APP - MQTT Disconnect. Reason 3
MQTT - Abort

g-oikonomou, 2017-04-24:
Hm yes this is likely related to keep-alive. I'll investigate.

balarsbp, 2017-05-02:
I am also facing same problem and some times sensor will goes to freezing stage(solid green state).in that time i restart the 6LBR or reboot the 6LBR still i cant recover it connection between 6lbr and sensor.

chenek, 2017-05-02:
@balarabp I think your issue is different. My sensor doesn't freeze.

balarsbp, 2017-05-02:
@chenek i set 600 sec interval for MQTT. I put continuously running 6LBR device and sensor .After 3 days sensor goes freezing condition.

alexandruioanp, 2017-06-25:
I don't currently have the means to connect to IBM quickstart. I ran the web demo for 24 hours with a local broker but I couldn't reproduce the issue.

chenek, 2017-06-26:
@alexandruioanp which issue can't you reproduce? Duration is set to 480 and you cannot see report in exactly 480 seconds? Or the hang issue?

alexandruioanp, 2017-06-26:
Sorry - I set the duration to 480 seconds and it reported at the correct interval.

balarsbp, 2017-06-27:
@alexandruioanp Are you use IBM MQTT server or Local server?

alexandruioanp, 2017-07-22:
Can you try https://github.com/contiki-os/contiki/pull/2263 and see if the problem still occurs?