contiki-os/contiki #2202: Cooja Mote Configuration Test Failures

adamrenner, 2017-04-21:
I have found a problem with the Cooja Mote Configuration Tests that is repeatable on several computers and releases.

Tests 3 and 4 of the test fail.
[image]
and
[image]

This occurs on release-3-0 and release-3-1, but not on release-2-7
Additionally, I do not experience these issues when doing this inside of Instant Contiki.

I suspect that this could be an issue related to 64 bit operating systems

I have also tried with several java versions including Sun 7 and 8 and OpenJDK 7 and 8.

Some information has suggested turning off extensions and ensuring the a 32 bit version of the ncurses library was installed, but these did not fix the problem.

sbungartz, 2017-04-26:
Hi, I'm getting the same results. However, for me this also happens in Instant Contiki 3.
From the looks of the Java code these tests can never succeed. For example in [ConfigurationWizard.java:763](https://github.com/contiki-os/contiki/blob/master/tools/cooja/java/org/contikios/cooja/dialogs/ConfigurationWizard.java#L763) we have:
```java
    addresses = new HashMap<String, Symbol>();
//    boolean parseOK = ContikiMoteType.parseMapFileData(mapData, addresses);
//    if (!parseOK) {
//      testOutput.addMessage("### Error: Failed parsing map file data", MessageList.ERROR);
//      return false;
//    }
```
and later, without doing anything else on `addresses` on [line 807](https://github.com/contiki-os/contiki/blob/master/tools/cooja/java/org/contikios/cooja/dialogs/ConfigurationWizard.java#L807):
```java
    String varName;
    varName = "var1";
    if (!addresses.containsKey(varName)) {
      testOutput.addMessage("Could not find address of: " + varName, MessageList.ERROR);
      return false;
    }
```
So these tests seem to be unmaintained.

In Instant Contiki 3 however, I successfully ran a simulation with a hello-world cooja mote.

Have you tried actually running a cooja mote? Sadly I did not have any success there either on my non-virtual system. I tried both gcc 5 and gcc 4 as well as Java 8 and Java 7.

Maybe it really is a 64 bit issue - I'm running 64 bit Ubuntu 16.04 and have only tried 64 bit jdks yet.

It would be nice if the developers could say something about the setups that cooja motes are currently expected to work with.

adamrenner, 2017-04-27:
Yes, I was running Cooja motes successfully in Instant Contiki, but Cooja would crash if I ran them in another operating system. That is what took me down the path of checking the results of the tests. I just assumed that the crashes and failures of the tests were related. This might be a bad assumption.

sbungartz, 2017-05-15:
Hey @adamrenner or anyone else who might be interested, I managed to work around this by creating a Docker image to run this from. Since Docker virtualization is much more lightweight, you don't really notice that it's not running on your host (which needs to be Linux though for this to be easy...).
I used a 32-bit Ubuntu 14.04 as the base system and it seems to work for me. (Might be worth trying 64-bit as well to see if that's actually the problem, but I only need something that works right now). You can check out the README at https://github.com/sbungartz/cooja-docker if you're interested.