contiki-os/contiki #2211: powertrace gnuplot script files are missing

anadiedrichs, 2017-05-03:
I have run on cooja a powertrace example, located at examples/powertrace. I have noticed some gnuplots scripts are missing, such as plot-node-power and plot-sniff-power, causing in errors for make powertrace*.

I run:

`make powertrace-all LOG=loglistener.txt`

Output:

`TARGET not defined, using target 'native'
cat loglistener.txt | grep -a "P " | ../../tools/powertrace/parse-power-data > powertrace-data
Node 8.0 avg duty cycle 1.9749970722118
Node 10.0 avg duty cycle 1.94513551315786
Node 9.0 avg duty cycle 1.90679824118561
Node 6.0 avg duty cycle 1.80591683157446
Node 3.0 avg duty cycle 1.98814192258744
Node 4.0 avg duty cycle 1.87979841218812
Node 2.0 avg duty cycle 1.89978167761841
Node 7.0 avg duty cycle 1.88524866366258
Node 5.0 avg duty cycle 2.00574405546579
Node 1.0 avg duty cycle 1.94523841868452
Mean duty cycle 1.93137124107071
cat loglistener.txt | grep -a "P " | ../../tools/powertrace/parse-node-power | sort -nr > powertrace-node-data
Idle percentage 0.332716322339557
cat loglistener.txt | ../../tools/powertrace/parse-sniff-data | sort -n > powertrace-sniff-data
Illegal division by zero at ../../tools/powertrace/parse-sniff-data line 47, <> line 1661.
Cannot open load file '../../tools/powertrace/plot-node-power'
"../../tools/powertrace/plot-node-power", line 0: util.c: No such file or directory

gnupot failed
Cannot open load file '../../tools/powertrace/plot-sniff-power'
"../../tools/powertrace/plot-sniff-power", line 0: util.c: No such file or directory

gnupot failed

set data style boxes
    ^
"../../tools/powertrace/plot-power", line 3: Unrecognized option.  See 'help set'.

gnupot failed
gv powertrace-power.eps
Warning: Cannot convert string "-*-Helvetica-Medium-R-Normal--*-140-*-*-P-*-ISO8859-1" to type FontStruct
Warning: Cannot convert string "-*-Helvetica-Medium-R-Normal--*-120-*-*-P-*-ISO8859-1" to type FontStruct
Warning: Cannot convert string "-*-Helvetica-Medium-R-Normal--*-100-*-*-P-*-ISO8859-1" to type FontStruct
gv: Cannot open file powertrace-power.eps (No such file or directory)`

anadiedrichs, 2017-05-03:
/contiki/tools/powertrace$ ls
Makefile.powertrace  parse-node-power  parse-power-data  parse-sniff-data  plot-power

The scripts plot-node-power and plot-sniff-power are not in the powertrace directory

Alchemisto, 2017-09-26:
I also have the same problem, did you manage to find the missing scripts?