contiki-os/contiki #2214: package javax.xml.bind is declared in module java.xml.bind, which is not in the module graph | Cooja Ant Build

kaleemullah360, 2017-05-06:
Latest clone causes trouble, please help. not interested in stable release

```
Buildfile: contiki/tools/cooja/build.xml

init:

compile:
    [javac] Compiling 170 source files to contiki/tools/cooja/build
    [javac] /contiki/tools/cooja/java/org/contikios/cooja/contikimote/interfaces/ContikiEEPROM.java:41: error: package javax.xml.bind is not visible
    [javac] import javax.xml.bind.DatatypeConverter;
    [javac]                 ^
    [javac]   (package javax.xml.bind is declared in module java.xml.bind, which is not in the module graph)
    [javac] Note: Some input files use or override a deprecated API.
    [javac] Note: Recompile with -Xlint:deprecation for details.
    [javac] Note: Some input files use unchecked or unsafe operations.
    [javac] Note: Recompile with -Xlint:unchecked for details.
    [javac] 1 error

BUILD FAILED
/home/ku/kontiki/tools/cooja/build.xml:77: Compile failed; see the compiler error output for details.

Total time: 7 seconds
```

kaleemullah360, 2017-07-08:
update JDK, JRE, solved the issue. thanks. closing it.

closed 2017-07-08 by kaleemullah360

hexiaole1994, 2017-10-06:
Hello,man
I have the same problem with yours
and I'm using jdk-9 which seems the newest version of jdk 
but still has problem,what should I do?
please help me

kilgore92, 2017-10-08:
This is an [issue with jdk-9](https://stackoverflow.com/questions/46084302/package-javax-xml-soap-is-declared-in-module-java-xml-ws-which-is-not-in-the-m). Install jdk-8 and set it to default. Should fix the problem.

hexiaole1994, 2017-10-08:
@kilgore92 Thank you,that will fix my issue 
and I wonder if it can be fixed at jdk-9
:0

Jean-Baptiste-B, 2017-10-12:
deprecated in JDK9 deprecated for removal, use JDK8 instead

JihenHlel, 2018-07-15:
@kaleemullah360  i had the same problem while runing ant, i updated jdk and jre but still the same error :( can you please give me the command line which you used for updating the jdk !

souhilahana, 2018-07-19:
@JihenHlel i have  the same problem!!!!

JihenHlel, 2018-07-19:
@souhilahana  already a week i posted my problem in multiples sites and forums but i didn't get any response :(

souhilahana, 2018-07-19:
I'm also, if you find a solution tell me please

Sambalinski, 2018-08-07:
https://www.logicbig.com/tutorials/core-java-tutorial/modules/root-modules.html

Vinggui, 2018-08-07:
Well, I think that using an old version of java is **not** the best solution.
Honestly, I'm not very experienced with ant and java, so I can not give a complete answer now.
### Immediate solution
Considering that you have ubuntu 16.04+ and installed all the necessary packages:
```
sudo apt-get install build-essential binutils-msp430 gcc-msp430 msp430-libc msp430mcu mspdebug gcc-arm-none-eabi openjdk-8-jdk openjdk-8-jre ant libncurses5-dev lib32ncurses5
```

Solving this problem is just:
```
sudo update-alternatives --config java
```
select the old java version (java-8-openjdk, probably option *2*)

### Better future solution
If someone else wants to solve this problem once and for all, a good start is in the build.xml file inside "tools/cooja".
Adding the following line made Cooja start and simulate:
`<compilerarg line="--add-modules java.xml.bind"/>`

Hence, the final compile command would be:
```
<target name="compile" depends="init">
    <mkdir dir="${build}"/>
    <javac srcdir="${java}" destdir="${build}" debug="on"
           includeantruntime="false"
           encoding="utf-8">
      <compilerarg line="--add-modules java.xml.bind"/>
      <classpath>
        <pathelement path="."/>
        <pathelement location="lib/jdom.jar"/>
        <pathelement location="lib/log4j.jar"/>
        <pathelement location="lib/jsyntaxpane.jar"/>
        <pathelement location="lib/swingx-all-1.6.4.jar"/>
      </classpath>
    </javac>
  </target>
```
However, some errors were shown, which did not stop simulation from working, so I don't know its impact.

**Machine tested:**
*Ubuntu 18.04 64bits
*java11(default) and java8 installed

souhilahana, 2018-08-26:
The Problem is solved thenkyouuuuuuu so much.
--

Souhila

gptshubham595, 2020-11-02:
I am using jdk 11

added it in as path in .bashrc
 also added 
```<compilerarg line="--add-modules java.xml.bind"/>```
But it says module not found

[image]

Vinggui, 2020-11-02:
@gptshubham595 I don't remember seeing this error. Also, you are using Contiki-ng. I don't know what changed in this way. Nevertheless, I'll suggest these steps to find the problem:
- Verify your Java installation and option with `sudo update-alternatives --config java`;
- Check if you installed the submodules correctly `git submodule update --init --recursive.`
Good luck. Please report back if you find the solution.