contiki-os/contiki #2218: Sleep mode in Contiki OS

saurabhbansal90, 2017-05-09:
I am using Contiki OS on CC2650. My requirement is to implement low power mode in it to increase the battery life. 
I enabled the maximum low power mode as LPM_MODE_DEEP_SLEEP. But whenever program enters in low power mode, it is exited by some timer event. I tried to change various parameters like MAX_SLEEP_TIME, MIN_SAFE_SCHEDULE, TSCH_CONF_CHANNEL_SCAN_DURATION, but there is same behaviour. It instantly comes out of sleep mode.

So I am not able to achieve long battery life on Contiki. Please let me know how I can reduce the waking up controller and any other method to achieve low power.

saurabhbansal90, 2017-05-11:
By looking into the code, I got to know RTC channel 0 is being used to schedule RF activities. And I am able to change their period of checking on RX by NETSTACK_CONF_RDC_CHANNEL_CHECK_RATE in contiki_conf.h file.
By default its value is set to 8, means it will check 8 times in a second. So maximum value can be kept to 1 sec. I changed its value to 1, and same thing is reflected by pinging also (the response coming as ~1sec).

And general timers like etimers seemed to be running from RTC channel 1, as in this channel is getting scheduled in lpm.c depending upon next etimer time.
For checking this, I commented RTC channel 1 scheduling in lpm.c, but I can't see any difference.

I want to sleep the controller for sometime (say 2sec), and in that period I don't want it to wakeup by any event. Can anybody tell me from where I can stop all event generation while sleep.

gxp1993, 2017-09-14:
Well, I just study Contiki one day....and I am not a native English speaker.....so its difficult to read my word.
You can take https://github.com/bkozak-scanimetrics/CC2650_lpm_test  for reference.
I use the Ti CC1310 chip and PCB is draw by myself.
I wonder how to wake up the chip from deep-sleep regularly.