contiki-os/contiki #2220: CC1310: Contiki OS - 6LoWPAN: RPL Hop Count

ijazfazily, 2017-05-10:
We have a design requirement to have 50 devices in a 6LoWPAN mesh network with linear positioning. We use MRHOF objective function and RPL storing mode.

With reference to our design requirements, Is there any stack limitation (other than memory space available on the device) on the maximum number of hops supported in mesh ?.  We would also like to know whether the hop count supported
by 6LoWPAN stack is configurable. If so, please let us know how to configure the same.
Please let us know if we shall provide you with any further information. Any suggestions/comments would be really appreciated.

mbaddeley, 2017-05-10:
I think this question is more suited to the mailing list rather than raising it as an issue. 

It depends on what data rate you need, how many of the nodes are generating messages, how often they are doing it, and what you set the route lifetime to. If you have a linear topology with every node sending a single packet to the sink at 10-20s intervals, and your route lifetime is set to 10 minutes, I **think** you would start to notice some performance degradation after about 15-20 hops. This is also assuming you are just sending data to the sink, and not responding to the individual motes (which involves the creation of downward routes, and is where you would start to hit limits on the hop count - depending on your hardware).

ijazfazily, 2017-05-11:
Thank you for your comments.
I tried mailing list and query got rejected.

In our design every node must send data packet to the border router at 5 minutes interval. And we have both up-link and down-link communication with route life time 1800 seconds. 

We would like to create a simulation in cooja for our design. Is there any configuration we need to set to have 50 devices. For e.g. "RPL_DAG_MC_HOPCOUNT" in rpl.h. 
We need to know the configurations required to support 50 hop count ( metric/constraint to be set ).

Regards,
Ijaz

lamnd09, 2017-06-20:
I think, RPL only supports ETX / Energy as metrics for routing. If you want to use hop count, maybe you need to use the ETX values and adapt to Hopcount values.