contiki-os/contiki #2221: cc2538dk: cannot flash with cc2538-bsl on current master

kYc0o, 2017-05-10:
Hi there!

I'm trying to flash any example on the contiki examples folder to a cc2538dk board. The error message is:

```console
python ../../tools/cc2538-bsl/cc2538-bsl.py -e -w -v -a 0x00200000 sensniff.bin
Opening port /dev/tty.usbserial-000022B, baud 500000
Reading data from sensniff.bin
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
Target returned: 0x42, Invalid command
ERROR: 'NoneType' object has no attribute '__getitem__'
make: *** [sensniff.upload] Error 1
rm sensniff.co obj_cc2538dk/startup-gcc.o
```

which gives me no clue about what is happening... any ideas?

g-oikonomou, 2017-05-10:
This is pretty strange, have submodules been updated to the latest version in your local copy?

kYc0o, 2017-05-10:
Yes, I updated everything, maybe it's an issue with the version of python or of the cc2538-bsl.

My python version:

```console
Python 2.7.10
```
For info I'm in a Mac OSX.

g-oikonomou, 2017-05-10:
I'm using the same version of Python, also on OS X, so it's gotta be something else. Can you try installing python-magic package and see what happens?

kYc0o, 2017-05-11:
Ok I will try that but it's a bit more strange since I'm using [this](https://github.com/RIOT-OS/RIOT/blob/master/dist/tools/cc2538-bsl/cc2538-bsl.py) script to flash it and it works correctly...

But if it works on your side I'm OK on closing the issue, maybe it's a more low level configuration or something related to drivers etc...

g-oikonomou, 2017-05-14:
Are you getting the same behaviour with the .bin files generated from other examples, e.g. the border router?

kYc0o, 2017-10-17:
Well I got into this test again and still have the issue (on Contiki-3.x-3148-geabb1ce, release 3.1). I replaced the script by the one I pointed out above and I can flash any Contiki example without any trouble...