contiki-os/contiki #2233: Modifying CC2650 sensortag IPv6 address

fbalseirolago, 2017-05-28:
Hi,

I have been working for some time now with 6lbr. My setup is basically a RPI2 which acts as a border router and in which 6lbr is deployed. The RPI2 has connected to one of its USB ports a CC2650 sensortag in which a slip-radio program has been flashed. The RPI2 will then communicate with other CC2650 sensors which run the cc26xx-web-demo through this 6LowPAN network deployed.

My issue comes when I want to replace my slip-radio with the following OpenLabs module: http://openlabs.co/store/Raspberry-Pi-802.15.4-radio . I have succesfully created a network interface and managed to ping two RPI2's through this module and interface however I am having problems when I try to ping one of the CC2650 sensortags from a RPI2. I think the main problem may reside in that the sensortags running the web-demo example come with a predefined IP address of fd00::/64 and therefore from my fe80::/64 interface of my RPI2 I am not able to achieve connectivity... (The sucesfull ping between the two RPI2's may occur because both interfaces have an IP address of fe80::/64)

Therefore my question: Is it possible to assign an fe80::/64 IPv6 address to the sensortags running the web-demo example? I think if I'm able to change their address I could achieve the connectivity wanted. If it is possible to change it, which lines of the source code should be modified?

Thank you very much and I hope the description above is clear.

bverdu, 2017-05-30:
Hello,
By default, contiki use the RPL protocol for the network discovery but there is no support for RPL in Linux, so if you want to achieve connectivity through the openlabs interface, you need to enable the classic IPV6 network discovery for your nodes in you project.conf file:
`#undef UIP_CONF_ND6_SEND_RA`
`#define UIP_CONF_ND6_SEND_RA 1
`

This is good enough to obtain local connectivity (with fe80::/64 address).

If you want to route the nodes to Internet you need to setup a router advertissement daemon (RADV) on the RaspberryPi.

Here is a complete guide: [https://github.com/RIOT-Makers/wpan-raspbian/wiki](https://github.com/RIOT-Makers/wpan-raspbian/wiki)

fbalseirolago, 2017-05-30:
Hi,
Thank you @bverdu for your quick reply and information !

So I did the changes you mention in the proj-conf.h file in the folder of cc26xx-web-demo (I am using this example due to the existence of the CoAP server that I will later need). 
From what I understand, if I was to declare
#define UIP_CONF_ND6_SEND_RA 1
I should be able to see router advertisements from my sensornode through wireshark... However I'm not sniffing any of these packages. 
Following your guidance, since RPL is not supported in Linux I have as well disabled the RPL solicitation messages from the sensor node, however still no luck... I still get destination unreachable message in my RPI when I try to ping the fe80:: address of the sensortag...
Do you have any ideas of why this may be ocurring? Do you think the fact that I don't see any RAs from my sensortag may be an issue? I tried understanding and reading the source code however I'm not able to see what other possible parameters I will have to change in order to send this RAs from the sensortag.

Thank you !

bverdu, 2017-05-30:
You have also to configure the same channel and Pan Id on both devices, the default values are not the same in Linux and Contiki, the command in Linux is iwpan.
You also need to disable ContkiMac in contiki and use nullrdc instead:
`#define NETSTACK_CONF_RDC   nullrdc_driver`
`#define NETSTACK_CONF_MAC   nullmac_driver`

you can try to ping the local mulicast address from the raspberryPi to check the connectivity:

`ping ff02::1%lowpan0`

fbalseirolago, 2017-06-01:
Thank you once again for your quick response !

Yes! The lines...
#define NETSTACK_CONF_RDC nullrdc_driver
#define NETSTACK_CONF_MAC nullmac_driver

managed to get the communication between the RPI and the sensortags going through the openalbs lowpan0 interface !! 

Thank you for yout help ! I will now close this issue

closed 2017-06-01 by fbalseirolago