contiki-os/contiki #2239: CoAP - memory corruption in coap_merge_multi_option (SIGSEGV)

stze, 2017-06-01:
Parsing the following coap message results in a crash of contiki:

coap message hexdump:
```
00000000  53 53 53 53 53 53 53 53  84 53 53 53 53 53 53 53  |SSSSSSSS.SSSSSSS|                          
00000010  5d 00 00 0e 53 53 53 73  32                       |]...SSSs2|
00000019
```
How to reproduce:
```c
static coap_packet_t message[1];
uint8_t teststr[] = { 0x53 ,0x53 ,0x53 ,0x53 ,0x53 ,0x53 ,0x53 ,0x53 ,0x84 ,0x53 ,0x53 ,0x53 ,0x53 ,0x53 ,0x53 ,0x53 ,0x5d ,0x00 ,0x00 ,0x0e ,0x53 ,0x53 ,0x53 ,0x73 ,0x32 };
coap_parse_message(message, (unsigned char *)teststr, sizeof(teststr));
```

gdb:
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f7b1837f49f in __memmove_avx_unaligned_erms () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007f7b1837f49f in __memmove_avx_unaligned_erms () from /usr/lib/libc.so.6
#1  0x000000000041502c in coap_merge_multi_option (dst=0x63a318 <message+216>, dst_len=0x63a310 <message+208>, option=0x7ffdf02a77c6 "2\377\377\377{\177", option_len=21600, separator=38 '&')
    at ../../apps/er-coap/er-coap.c:231
#2  0x0000000000414ad0 in coap_parse_message (packet=0x63a240 <message>, data=0x7ffdf02a77b0 "SSSSSSSS\204SSSSSSS]", data_len=25) at ../../apps/er-coap/er-coap.c:621
#3  0x00000000004077e2 in process_thread_er_example_client (process_pt=0x639198 <er_example_client+24>, ev=<optimized out>, data=<optimized out>) at er-example-client.c:115
#4  0x7830000000000041 in ?? ()
#5  0x000000007ffdf02a in ?? ()
#6  0x0008000000000000 in ?? ()
#7  0x0000000000000000 in ?? ()
```
valgrind:
```
==18107== Invalid read of size 2
==18107==    at 0x4C31A08: memmove (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==18107==    by 0x41502B: coap_merge_multi_option (er-coap.c:231)
==18107==    by 0x414ACF: coap_parse_message (er-coap.c:621)
==18107==    by 0x4077E1: process_thread_er_example_client (er-example-client.c:115)
==18107==    by 0x310000000000040: ???
==18107==    by 0xFFEFF: ???
==18107==    by 0x7FFFFFFFFFFFF: ???
==18107==  Address 0xfff001000 is not stack'd, malloc'd or (recently) free'd
==18107== 
==18107== 
==18107== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==18107==  Access not within mapped region at address 0xFFF001000
==18107==    at 0x4C31A08: memmove (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==18107==    by 0x41502B: coap_merge_multi_option (er-coap.c:231)
==18107==    by 0x414ACF: coap_parse_message (er-coap.c:621)
==18107==    by 0x4077E1: process_thread_er_example_client (er-example-client.c:115)
==18107==    by 0x310000000000040: ???
==18107==    by 0xFFEFF: ???
==18107==    by 0x7FFFFFFFFFFFF: ???
```

The crash has been found with afl-fuzz.

Best Regards,
Stephan Zeisberg

simonduq, 2017-06-02:
Thanks for reporting!

nvt, 2017-06-15:
PR #2249 has been merged to fix this issue. Thanks!

closed 2017-06-15 by nvt