contiki-os/contiki #2247: er-rest-example compile failed

Windywind, 2017-06-13:
Default example cloned from git compile failed (but the example in Instant Contiki works)

[image]

Command line output:
```
$ make TARGET=cooja server-client.csc
java -mx512m -jar ../../tools/cooja/dist/cooja.jar -quickstart='server-client.csc' -contiki='../..'
 WARN [main] (Cooja.java:3172) - JAVA_HOME environment variable not set, Cooja motes may not compile
 INFO [main] (Cooja.java:1315) - > Starting Cooja
 INFO [main] (Cooja.java:2856) - External tools default settings: /external_tools_linux_64.config
 INFO [main] (Cooja.java:2886) - External tools user settings: /home/wind/.cooja.user.properties
 INFO [main] (Simulation.java:422) - Simulation random seed: 123456
 WARN [main] (Cooja.java:4261) - Replaced [CONTIKI_DIR]/examples/ipv6/rpl-border-router/border-router.sky with /home/wind/contiki/examples/ipv6/rpl-border-router/border-router.sky (default: ../..), but could not find it. This does not have to be an error, as the file might be created later.
 INFO [main] (CompileContiki.java:140) - > make border-router.sky TARGET=sky 
FATAL [main] (SkyMoteType.java:171) - >> /home/wind/contiki/examples/ipv6/rpl-border-router/../../../platform/sky/./contiki-sky-main.c:287: relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
FATAL [main] (SkyMoteType.java:171) - >> /home/wind/contiki/examples/ipv6/rpl-border-router/../../../platform/sky/./contiki-sky-main.c:287: relocation truncated to fit: R_MSP430_16 against symbol `puts' defined in .text section in /usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mmpy-16/libc.a(puts.o)
FATAL [main] (SkyMoteType.java:171) - >> /home/wind/contiki/examples/ipv6/rpl-border-router/../../../platform/sky/./contiki-sky-main.c:302: relocation truncated to fit: R_MSP430_16 against symbol `contikimac_driver' defined in .rodata section in contiki-sky.a(contikimac.o)
FATAL [main] (SkyMoteType.java:171) - >> /home/wind/contiki/examples/ipv6/rpl-border-router/../../../platform/sky/./contiki-sky-main.c:303: additional relocation overflows omitted from the output
FATAL [main] (SkyMoteType.java:171) - >> collect2: ld returned 1 exit status
FATAL [main] (SkyMoteType.java:171) - >> make[1]: *** [border-router.sky] Error 1
FATAL [main] (SkyMoteType.java:171) - >> ../../../Makefile.include:280: recipe for target 'border-router.sky' failed
FATAL [main] (SkyMoteType.java:171) - >> rm border-router.co obj_sky/contiki-sky-main.o
FATAL [main] (SkyMoteType.java:171) - >> make[1]: Leaving directory '/home/wind/contiki/examples/ipv6/rpl-border-router'
FATAL [main] (SkyMoteType.java:171) - >> Process returned error code 2
FATAL [main] (SkyMoteType.java:174) - Compilation error: Bad return value
$ make TARGET=cooja server-client.csc
java -mx512m -jar ../../tools/cooja/dist/cooja.jar -quickstart='server-client.csc' -contiki='../..'
 WARN [main] (Cooja.java:3172) - JAVA_HOME environment variable not set, Cooja motes may not compile
 INFO [main] (Cooja.java:1315) - > Starting Cooja
 INFO [main] (Cooja.java:2856) - External tools default settings: /external_tools_linux_64.config
 INFO [main] (Cooja.java:2886) - External tools user settings: /home/wind/.cooja.user.properties
 INFO [main] (Simulation.java:422) - Simulation random seed: 123456
 WARN [main] (Cooja.java:4261) - Replaced [CONTIKI_DIR]/examples/ipv6/rpl-border-router/border-router.sky with /home/wind/contiki/examples/ipv6/rpl-border-router/border-router.sky (default: ../..), but could not find it. This does not have to be an error, as the file might be created later.
 INFO [main] (CompileContiki.java:140) - > make border-router.sky TARGET=sky 
FATAL [main] (SkyMoteType.java:171) - >> /home/wind/contiki/examples/ipv6/rpl-border-router/../../../platform/sky/./contiki-sky-main.c:287: relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
FATAL [main] (SkyMoteType.java:171) - >> /home/wind/contiki/examples/ipv6/rpl-border-router/../../../platform/sky/./contiki-sky-main.c:287: relocation truncated to fit: R_MSP430_16 against symbol `puts' defined in .text section in /usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mmpy-16/libc.a(puts.o)
FATAL [main] (SkyMoteType.java:171) - >> /home/wind/contiki/examples/ipv6/rpl-border-router/../../../platform/sky/./contiki-sky-main.c:302: relocation truncated to fit: R_MSP430_16 against symbol `contikimac_driver' defined in .rodata section in contiki-sky.a(contikimac.o)
FATAL [main] (SkyMoteType.java:171) - >> /home/wind/contiki/examples/ipv6/rpl-border-router/../../../platform/sky/./contiki-sky-main.c:303: additional relocation overflows omitted from the output
FATAL [main] (SkyMoteType.java:171) - >> collect2: ld returned 1 exit status
FATAL [main] (SkyMoteType.java:171) - >> make[1]: *** [border-router.sky] Error 1
FATAL [main] (SkyMoteType.java:171) - >> ../../../Makefile.include:280: recipe for target 'border-router.sky' failed
FATAL [main] (SkyMoteType.java:171) - >> rm border-router.co obj_sky/contiki-sky-main.o
FATAL [main] (SkyMoteType.java:171) - >> make[1]: Leaving directory '/home/wind/contiki/examples/ipv6/rpl-border-router'
FATAL [main] (SkyMoteType.java:171) - >> Process returned error code 2
FATAL [main] (SkyMoteType.java:174) - Compilation error: Bad return value

```

Windywind, 2017-06-13:
I can see this is because of boarder-router will not compile for sky mote. I tried to compile it alone, and it won't compile as expected. 
The output is similar to that of above.

nvt, 2017-06-15:
This is a request for help rather than a Contiki issue, so I'm closing it. But I can quickly mention that if you only intend to run this in Cooja, you can use the Wismote instead. It is similar to the Sky platform, but has considerably more ROM space. You'll need a newer compiler to be able to use it -- for example mspgcc 4.7.2.

Otherwise you would have to tune the firmware size. https://github.com/contiki-os/contiki/wiki/Reducing-Contiki-OS-firmware-size provides some suggestions on how to do this.

closed 2017-06-15 by nvt

Windywind, 2017-07-23:
Hi @nvt ,
I am desperately want to compile msp430-gcc on my own machine, but the script provided in wiki doesn't work.

Link: [https://gist.github.com/madmo/3544544](url)

I have looked into it, even the 'install-prefix' is set to '/tmp/'(not as described '/opt/') After corrected this, it still will not compile through. 

So, could you give some suggestion? How can I get a msp430-gcc 4.7.0 version?

Thanks.