contiki-os/contiki #2248: A fatal error has been detected by the Java Runtime Environment: SIGSEGV (0xb) at pc=0x00007f70ec7ff009, pid=4165, tid=4197

RamziFsm, 2017-06-13:
Hi,
I installed Contiki 3.0 under Ubuntu 16.04 64 bits.I can't start cooja simulator.I used Oracle JDK 8 but errors are generated. Here is the result after building: 

run:

[java] Exception in thread "AWT-EventQueue-0" java.lang.BootstrapMethodError: java.lang.NoClassDefFoundError: java/lang/invoke/StringConcatFactory

[java] at org.contikios.cooja.Cooja.loadQuickHelp(Cooja.java:4417)

[java] at org.contikios.cooja.Cooja.(Cooja.java:385)

[java] at org.contikios.cooja.Cooja$30.run(Cooja.java:3331)

[java] atjava.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)

[java] at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)

[java] at java.awt.EventQueue.access$500(EventQueue.java:97)

[java] at java.awt.EventQueue$3.run(EventQueue.java:709)

[java] at java.awt.EventQueue$3.run(EventQueue.java:703)

[java] at java.security.AccessController.doPrivileged(Native Method)

[java] at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)

[java] at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)

[java] at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)

[java] at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)

[java] at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)

[java] at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)

[java] at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)

[java] at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

[java] Caused by: java.lang.NoClassDefFoundError: java/lang/invoke/StringConcatFactory

[java] ... 17 more

[java] Caused by: java.lang.ClassNotFoundException: java.lang.invoke.StringConcatFactory

[java] at java.net.URLClassLoader.findClass(URLClassLoader.java:381)

[java] at java.lang.ClassLoader.loadClass(ClassLoader.java:424)

[java] at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:335)

[java] at java.lang.ClassLoader.loadClass(ClassLoader.java:357)

[java] ... 17 more

BUILD SUCCESSFUL
I am not able to understand what is the issue. Can anyone tell me how to resolve it? thanks,

Windywind, 2017-06-14:
Follow instructions strictly
https://github.com/contiki-os/contiki/wiki/Setup-contiki-toolchain-in-ubuntu-13.04

ssm12, 2017-12-21:
Hi, 
Even I got the same error. Try running below commands.

1. sudo add-apt-repository ppa:openjdk-r/ppa

2.After that, update system package cache and install OpenJDK 8: run below command
 sudo apt-get update 

sudo apt-get install openjdk-8-jdk
3.If you have more than one Java versions installed on your system. Run below command set the default Java:
sudo update-alternatives --config java

select the corresponding option shown in the prompt after running the above command. Say select 1 or 2 or 3 based on the version you are interested to use.Example - 1: may contain openjdk-7,  2: may contain openjdk-8 etc.
And set default Java Compiler by running:
sudo update-alternatives --config javac
4. . Finally check out current Java version by running:
java -version
It outputs something like this:
openjdk version "1.8.0_01-internal"
OpenJDK Runtime Environment (build 1.8.0_01-internal-b04)
OpenJDK 64-Bit Server VM (build 25.40-b08, mixed mode)

source : http://ubuntuhandbook.org/index.php/2015/01/install-openjdk-8-ubuntu-14-04-12-04-lts/