contiki-os/contiki #2253: Multithreading support for CC1350 in Contiki

pritibakale, 2017-06-20:
Hello,

I am trying to combine unicast-sender and unicast-receiver code to achieve bidirectional communication using UDP. But I was facing few problems in doing so. At a time only one of the send or receive function was working properly. Now I am using Multi threading along with the combined code. But when I tried to compile and build the .bin file for CC1350 platform it gave following errors

contiki-srf06-cc26xx.a(mt.o): In function `mt_init':
mt.c:(.text.mt_init+0x0): undefined reference to `mtarch_init'
contiki-srf06-cc26xx.a(mt.o): In function `mt_start':
mt.c:(.text.mt_start+0x6): undefined reference to `mtarch_start'
contiki-srf06-cc26xx.a(mt.o): In function `mt_exec':
mt.c:(.text.mt_exec+0x10): undefined reference to `mtarch_exec'
contiki-srf06-cc26xx.a(mt.o): In function `mt_yield':
mt.c:(.text.mt_yield+0x2): undefined reference to `mtarch_pstop'
mt.c:(.text.mt_yield+0xa): undefined reference to `mtarch_yield'

Is CC1350 is not supported platform for Multi threading. If not then how can I achieve multitasking? 
If supported then what changes to be done to remove reeor?

Kindly suggest me solution.

alexrayne, 2017-08-02:
contiki net stack not ddesigned for concurent multi-threading. even if you standup concurent sheduler, net routines not expects about unexpected switching, on concurent reentant run. 
Look for protothreads. that`s why network API so triky - rime connection takes callbacks set for handle sent, receive events.
for concurent send, receive work just look for example-neighbors

vamsi-chikati, 2017-08-02:
As HW progressed and MCUs evolved with relatively large RAMs, it feels like
a natural next step for Contiki to adapt general RTOS execution model with
multi-threading as the default than a special case.
Though the proto-threads are indeed  clever, and  innovative they are not
intuitive for regular programmers. Thus the learning curve is steep.
Reading code is tough too. To put it mildly, it is a compromise to save ram
when it is not that scarce anymore.

msloth, 2017-09-04:
Protothreads can be tricky. The `PROCESS` abstraction is more straight-forward and you come a long way using that.

There are drawbacks with pre-emptive multithreading too, of course. Contiki is using cooperative multithreading - each (proto)thread gives back control to the scheduler. Just use the common abstractions (`PROCESS_WAIT_EVENT_UNTIL` etc) and you'll be fine; check eg `process.h`. Get familiar with them - like with all systems you need to know what you are using. There are plenty of tutorials and guides.

This question is better suited at eg stackoverflow, tag it with Contiki.