contiki-os/contiki #2268: CoAP Forward Proxy implementation？

Windywind, 2017-07-19:
This is a question about the CoAP implementation. 

Is 'forward proxy' implemented in ContikiOS, or not? Or, say Cooja, I am using Cooja for simulation.

To make it easier to disscuss, let's say, we have such a senario:

                                            C——A——B
- B as a server, has the resources, which C is going to request.
- A is an intermediate node

Q1: If A is an RPL border router, will it be able to forward? (In Cooja and in real life)
Q2: If A is a CoAP node, will it be able to forward?(In Cooja and in real life)
Q3: If needed, is there any setup have to be down? (e.g. set RPL to storing, non-storing?)

Thanks a lot!

aichaab, 2018-04-22:
hy Windywind, actualy I have the same probleme ....

Windywind, 2018-04-24:
Hi @aichaab , I thought and realized that a IoT node in any cases needs a routing table and a neighbor table to do all of these. So, please try to find out a solution with this kept in mind.

aichaab, 2018-04-24:
thanks @Windywind  ,I really apreciate yourhelp. Actualy i'm working on a third party C authentication wich play a rol of a gateway wich have two roles (client and server), C needs to initiate two sessions  one with B and the auther with A at the same time , and this what makes problem in my work .