contiki-os/contiki #2270: CC2650: Can't access CC2650-web-demo node web page

BoKarolis, 2017-07-31:
Hi, I'm new to all this Contiki-6lbr thing. My setup is: 6lbr develop-20170121 running on a recently updated raspberry pi 2B with a Wi Pi dongle. I'm using srf06eb+cc2650em-7id as a slip radio (followed [this](http://sunmaysky.blogspot.de/2015/09/how-to-build-contiki-6lbr-slip-radio-on.html) guide) and another srf06eb+cc2650em-7id flashed with a cc26xx-web-demo. I am able to ping and access my 6lbr router webpage using my raspberry pi, the other board is shown in the sensors list, hence it connects (the LED on the cc26xx-web-demo board starts blinking slowly) and it sends the UDP packets, seen in the log, BUT I cannot access neither the web page served by the web-demo node nor the CoAP. I have tried various things such as: 

- Changing 6lbr/Contiki versions between develop and stable/master branches
- Setting PAN-ID to 0xabcd and channel to 25/26/13 on slip-radio project-conf.h
- Reinstalling, redownloading, rebuilding and reflashing everything
- Swapping to a BBB
- Switching to RAW_ETH mode and trying to access the web node via Android 7.0 tablet
And the same issue persists, I can open the router web page, but not the node.

My 6lbr.conf looks like this: 
```
MODE=ROUTER
RAW_ETH=0
BRIDGE=1
CREATE_BRIDGE=0
DEV_BRIDGE=br0
DEV_TAP=tap0
DEV_ETH=wlan0
RAW_ETH_FCS=0
DEV_RADIO=/dev/ttyUSB1
BAUDRATE=115200
LOG_LEVEL=4
```
my /etc/network/interfaces looks like this: 
```
auto lo
iface lo inet loopback
allow-hotplug wlan0
iface wlan0 inet manual
wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf
auto br0
iface br0 inet static
    bridge_ports wlan0
    bridge_stp off
    up echo 0 > /sys/devices/virtual/net/br0/bridge/multicast_snooping
    post-up ip link set br0 address `ip link show wlan0 | grep ether | awk '{print $2}'`
    address 192.168.2.100
    netmask 255.255.255.0
    network 192.168.2.0
    broadcast 192.168.2.255
    gateway 192.168.2.254
iface default inet dhcp
```
I am not sure if this is related to the issue, but in the 6lbr.log I get errors notifying 
```
ERROR: BR-RDC: Set channel failed
ERROR: BR-RDC: Set PAN-ID failed
```
However I tried setting those things at slip-radio project-conf.h using 
```
#define IEEE802154_CONF_PANID            0xABCD
#define RF_CORE_CONF_CHANNEL                 13
```
and it did not help. I have been struggling with this issue for a few days now, read through numerous posts, but I could not solve this issue. Any help will be hugely appreciated. Thank you for your time.

BoKarolis, 2017-08-02:
Update: After messing around with the specific configurations for raspberry pi, I got access to the node webpage. Another problem is that I can just use the raspberry to access it. Any other devices cannot connect neither to the 6lbr webpage nor the node webpage. If I use RAW_ETHERNET=1 mode then I can access the router webpage from other devices on the network, but still cannot open node's webpage.

jacekrz, 2017-09-12:
Hi
I have had his issue (MODE=ROUTER, BRIDGE=1) . My solution is to add to router table (pc as a client) line:
in Linux - temporarily
route -A inet6 add fd00::/64 gw bbbb::100 dev wlan0
where 'fd00::/64' is your 6LowPAN and bbbb::100 ipv6 your Raspi
in Windows - persistant
netsh interface ipv6 add route ffd0::/64 "Your_network_interface" bbbb::100
where Your_network_interface is wlan0, eth0 ...
I assume you can ping6 to bbbb::100.
Regards

closed 2017-10-26 by BoKarolis