contiki-os/contiki #2280: How to resolve msp430 problem?

iamxg, 2017-08-04:
[image]

iamxg, 2017-08-04:
msp430-gcc 4.6 upgrade to 4.7, but this issue still occur.

fabricio-godoi, 2017-08-04:
@iamxg if you check carefully the line highlight the error mention that the code you trying to compile it's too big for the Sky mote (overflowed by 26KB of ROM).
I've seen a lot of people with this same issue. I recommend you to start with the Wismote, which has a lot more memory to run codes.

If you strictly need the Sky mote, I sugest you take a moment and reanalyze all specification needed in your project.

By the way, this is the motes memory:
Sky 48 KB ROM
Z1 up to 96 KB ROM 
Wismote up to 256 KB ROM

Note: check if the makefile is configured correctly to use the memory needed.

Best regards!

iamxg, 2017-08-04:
Thanks a lot.

Happy weekend,

 

Xian G

lamnd09, 2017-08-05:
4.7.2 version works well, you can refer to past thread about this issue.

iamxg, 2017-08-06:
Ok, Thank you very much! I will try it.

iamxg, 2017-08-06:
[image]

iamxg, 2017-08-06:
It can't work.

lamnd09, 2017-08-06:
it's much better if you use Z1 instead of Sky.

pcumino, 2017-08-09:
I'm having a similar issue. In my case I can compile a **skymote** as a border router using the file contiki/examples/ipv6/rpl-border-router/border-router.c (just as shown in [here](http://anrg.usc.edu/contiki/index.php/RPL_Border_Router).
But when I try to use a **wismote** instead the error comes up. It says the there is no ROM available.
I've already tried:
To use the msp430-gcc v4.7.0 (in this case the simulator didn't open);
To use two different versions installed msp430-gcc (4.6.3 and 4.7.0), but the error still occur.

iamxg, 2017-08-09:
Hi everyone, Thanks a lot. I will try these methods.

fabricio-godoi, 2017-08-10:
One last hint I can give you guys is to check if the Makefiles are correct.
Each platform has it's own makefile, like Makefile.wismote or Makefile.z1, look in it and check if it's enabled the CPU_HAS_MSP430X=1. This will enable in Makefile.msp430 to configure the ROM size of the MCU.
Note that Sky mote does not have this feature!

Also, check the "application".map file, there will be some section like this below, so check the actual length of the ROM to see how much is needed to run the code. Note that it's value is absolutely after compiling the code:
`Memory Configuration

Name             Origin             Length             Attributes
sfr              0x00000000         0x00000010
peripheral_8bit  0x00000010         0x000000f0
peripheral_16bit 0x00000100         0x00000100
bsl              0x00001000         0x00000800
infomem          0x00001800         0x00000200
infod            0x00001800         0x00000080
infoc            0x00001880         0x00000080
infob            0x00001900         0x00000080
infoa            0x00001980         0x00000080
ram              0x00001c00         0x00004000         xw
rom              0x00005c00         0x0000a380         xr
vectors          0x0000ff80         0x00000080
far_rom          0x00010000         0x00035c00
ram2             0x00000000         0x00000000         xw
ram_mirror       0x00000000         0x00000000         xw
usbram           0x00000000         0x00000000         xw
*default*        0x00000000         0xffffffff`

For more information see the options of the msp430-gcc:
https://gcc.gnu.org/onlinedocs/gcc/MSP430-Options.html

/platform/Wismote/Makefile.wismote
/cpu/msp430/Makefile.msp430

By the way, I'm using msp430-gcc 4.7.0.

Good luck!
Best regards!

atiselsts, 2017-08-11:
@fabricio-godoi program code is also stored in `far_rom` section - that's the one `sky` is missing and msp430-gcc-4.6 does not support.
If `CPU_HAS_MSP430X=1` is defined and the toolchain is recent enough, it will put some parts of code in addresses >= 0x00010000, i.e. the extended memory.

Rumor has it that 4.7.2 and 4.7.3 has fixed some bugs related to the extended memory support.
Get the binaries here: https://github.com/pksec/msp430-gcc-4.7.3

pcumino, 2017-08-15:
I solved the problem by updating the compiler following the instructions in [here](https://github.com/pksec/msp430-gcc-4.7.3).
The main steps I followed:
  1) Installing msp430-gcc version **4.7.3** and restarted the VM. So both versions are installed. The shortcut for the new version is _msp430-gcc-4.7.3_ 
  2)  Changing the file **contiki/cpu/msp430/Makefile.msp430** . Updating the compiler used in CC and LD lines to the new one.

```
#CC    = msp430-gcc
CC     = msp430-gcc-4.7.3
#LD    = msp430-gcc
LD     = msp430-gcc-4.7.3
```
Now it is possible to compile the rpl-border-router.c using the wismote platform with 20-bit support.

Thanks everyone.

vishal623, 2017-11-08:
Hello,

I followed the same steps as per mentioned here but when I compiled with any mote like sky, z1 or wisemote it gives me the following error.

make: msp430-gcc-4.7.3: Command not found
fatal: Not a valid object name HEAD

Please help me.

iamxg, 2017-11-08:
Sorry, I don't know how to resolve. My problem also is not resolved until now.

 

 Regards,

Xian G

fabricio-godoi, 2017-11-16:
Hello everyone,

I reach a similar problem today, even with the CPU_HAS_MSP430X=1,
aparrently there some bug with this directive, not sure though.

I had to change the mcode-region flag from 'any' to 'far' at Makefile.msp430 line 156,
and then the code was able to be compiled.

Before:
  CFLAGS += -ffunction-sections -fdata-sections -mcode-region=any
After:
  CFLAGS += -ffunction-sections -fdata-sections -mcode-region=far

@atiselsts 
I didn't know about the GCC versions problems, I just assumed that was 
some kind of flag configuration problem, thanks you about it.
The sky mote does not have this feature because it has only 48K of ROM,
so I recommended switch it to the Wismote.

Regards

greg-king5, 2017-11-21:
@vishal623, you must point your `PATH` environment variable to the 4.7.3 version's "bin" directory (step `#2`).  But, don't do it in "`~/.bashrc`"!  Do it in "`~/.bash_profile`".

vishal623, 2017-11-23:
Thanks, sir. I will try and let you know its working for me or not.

Siddarth1511, 2018-04-10:
was this problem solved ??

And by the way @vishal623 I had the same problem,
am still having troubles in upgrading my msp gcc compilers did you solve it?

vishal623, 2018-04-10:
Yes. It is working for me. When I have checked the version of msp430, it
shows me the new version, but Contiki-3.0 doesn't accept it for me.

Siddarth1511, 2018-04-10:
what did you do to solve it @vishal623 ?
I changed path in .profile it shows me this 
[image]

vishal623, 2018-04-10:
I just switched back to Contiki-2.7 and all is well.

vishal623, 2018-04-10:
> oh alright
> i will try doing that
> did you follow the 2 steps given in the readme file
> and also did you change CC = msp430-gcc line in makefile
>
> @vishal623 <https://github.com/vishal623>

Yes.

greg-king5, 2018-04-11:
Try changing the PATH line to:
```sh
PATH=/opt/compilers/mspgcc-4.7.3/bin:$PATH
```

Siddarth1511, 2018-04-11:
In which file .bashrc or .profile?? @greg-king5 
Thanks for helping me out

greg-king5, 2018-04-12:
`~/.bashrc` is for things that aren't inherited by shells -- such as aliases, environment variables that aren't exported, and functions.  `~/.profile` (or `~/.bash_profile`) is for system settings and exported variables (such as `PATH`).

Siddarth1511, 2018-04-13:
It worked!!!
Thank you so much @greg-king5 @vishal623 
It didn't work when i tried what you said in Instant contiki (VMware player)
But worked when i tried in dual boot ubuntu 14.04
Thanks again!

tejaeluri, 2018-04-29:
@greg-king5  @Siddarth1511 shoud i have to remove msp430-gcc {4.6.3} packages ? before doing the above steps? updating to 4.7.3 ? please help...working on contiki...er-coap..ubuntu 16.04

Siddarth1511, 2018-04-29:
are you using instant contiki?
or a dual boot ubuntu os with contiki?
@tejaeluri

tejaeluri, 2018-04-29:
Dual boot Ubuntu 16.04 LTS with contiki 3.0

Siddarth1511, 2018-04-29:
Alright
i didn't remove anything
just extracted the files to the location, changed the compiler version in makefile.msp430, added the path in .profile 
restarted the system
worked like a charm

tejaeluri, 2018-04-29:
ohh...thanks...can you tell me which versions of ubuntu and contiki you are using??

Siddarth1511, 2018-04-29:
ubuntu 14.04 and i was using contiki-master

tejaeluri, 2018-04-30:
"add PATH MSP430/bin path to your environment " i didnt get this
point..should i have to add path in /etc/environment ?? Please explain
this..
@Siddarth1511

tejaeluri, 2018-05-01:
@pcumino @greg-king5 @Siddarth1511 Guyz..help me out..how to add path in environment ?? im following these instructions... [https://github.com/pksec/msp430-gcc-4.7.3](url)

Siddarth1511, 2018-05-01:
you have to open the .profile file ( gedit .profile) and add wherever you placed your msp430(4.7.3) folder 
Then go to makefile.msp430 change CC and LD to msp430-gcc-4.7.3 
restart your system and check it

Sara-94, 2019-03-20:
Hello @vishal623  
I also follow the same steps and i have the same problem, the version shows me 4.7.3 but Cooja shows me msp430-gcc-4.7.3 command not found.
please did you solve the problem?

Sara-94, 2019-03-20:
Thank you so much for replay @vishal623 
This is the echo of the PATH 
[image]
it point to 4.7.3 version

Sara-94, 2019-03-20:
Thank you so much for replay @Siddarth1511
This is the echo of the PATH
[image]
it point to 4.7.3 version

vishal623, 2019-03-21:
Even I have the same error since last 8 months.

Sara-94, 2019-03-21:
Thanks for your replay @vishal623 
I tried from 3 weeks and try the steps for Contiki 3.0 and Contiki 2.7, also i try it in other computer and its still not work
Is there any other ways to increase the memory of z1 mote?
Note: i know that the wismote have more memory than z1 but when i try the code on z1 it overflow by 2300 bytes (for coap server and border router run without problem) and when try wismote it overflow by 7476 bytes (for border router) and 12982 (for coap server), why the wismote seems to have less memory than z1 mote??

Sara-94, 2019-03-25:
Hello @vishal623 
I can force Cooja to use msp430-gcc-4.7.3 by using  "sudo update-alternatives --install /usr/bin/msp430-gcc gcc /opt/compiler/mspgcc-4.7.3/bin/msp430-gcc-4.7.3 50"
also use "sudo update-alternatives --install /usr/bin/msp430-gcc gcc /usr/bin/msp430-gcc-4.6.3 40" so you can choose any version you want by using  "sudo update-alternatives --config gcc" 
but another error appear as this picture 
[image]

vishal623, 2019-03-25:
Let me try this and let you know about the same.

tejaeluri, 2019-03-26:
This is what i did (it works for me):
1- install ubuntu 14-04 (32bit)
2- download latest contiki 3.0 from sourceforge.net
2.1- extract contiki 3.0 files in "/home/"
3- Download the msp430-47 binary files from sourceforge.net
(*msp430-47 is an experimental tool chain which gives extra 20bit memory for motes)
3.1- Extract the files in "/home/msp430-47"
3.2- In Home directory click (Ctrl+H) to open hidden files and find ".bashrc" and ".profile" files
3.3 open those files and add “export PATH=/home/user/msp430-47/bin:$PATH” in end of the codes.
(*this Path will helps the Contiki to use the tool chain)
4- To increase Z1 motes memory
4.1 go to path "Home/contiki/platform/z1"
4.2 open file "contiki-conf.h"
4.3 increase buffer size to 240
" #define UIP_CONF_BUFFER_SIZE    240" and save it
4.4 restart pc 
4.5 its done

vishal623, 2019-03-26:
When I tried following command

sudo update-alternatives --install /usr/bin/msp430-gcc gcc
/opt/compiler/mspgcc-4.7.3/bin/msp430-gcc-4.7.3 50

it shows me following error in cooja:

msp430-gcc: error: unrecognized command line option ‘-mmcu=msp430f2617’

greg-king5, 2019-03-27:
> 3.2- In Home directory, click (Ctrl+H) to open hidden files, and find ".bashrc" and ".profile" files.
> 3.3 Open those files, and add “export PATH=/home/user/msp430-47/bin:$PATH” in end of the codes.
> (*that Path will help the Contiki to use the tool chain.)

Don't put that "export PATH ..." line in `"~/.bashrc"`.  Put it in only `"~/.profile"`.

vishal623, 2019-03-27:
I already added the path in ~/.profile file.

When I typed gcc --version it shows me 4.7.3 but when I typed msp430-gcc
--version it shows me 5.5.0. See below output.

vishal@vishal-OptiPlex-9020:~$ gcc --version

gcc (GCC) 4.7.3 20130411 (mspgcc dev 20120911)
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

vishal@vishal-OptiPlex-9020:~$ msp430-gcc --version

msp430-gcc (Ubuntu 5.5.0-12ubuntu1) 5.5.0 20171010
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

This happened after executing the "sudo update-alternatives --install
/usr/bin/msp430-gcc gcc /opt/compiler/mspgcc-4.7.3/bin/msp430-gcc-4.7.3 50"
this command.

Why?

greg-king5, 2019-03-27:
> ```
> vishal@vishal-OptiPlex-9020:~$ gcc --version
> gcc (GCC) 4.7.3 20130411 (mspgcc dev 20120911)
> Copyright (C) 2012 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.
> There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> 
> vishal@vishal-OptiPlex-9020:~$ msp430-gcc --version
> msp430-gcc (Ubuntu 5.5.0-12ubuntu1) 5.5.0 20171010
> Copyright (C) 2015 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.
> There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> ```
> That happened after executing the `"sudo update-alternatives --install /usr/bin/msp430-gcc gcc /opt/compiler/mspgcc-4.7.3/bin/msp430-gcc-4.7.3 50"` command.  Why?

The `gcc` behavior looks as though you already have a "gcc" group in your alternatives.  That `update-alternatives` command added another alternative link that has a higher priority than the one that `gcc` should use.  You must remove the new link:
```sh
sudo update-alternatives --remove gcc /opt/compiler/mspgcc-4.7.3/bin/msp430-gcc-4.7.3
```

The `msp430-gcc` behavior looks as though your PATH points to the old compiler's "bin/" before it points to "/usr/bin/".  The "alternatives" method cannot work for you.

Is there a line in your "~/.profile" that points to the old compiler's "bin/"?  If so, then comment it.

vishal623, 2019-03-28:
Now, I am able to get the 4.7.3 for both.

vishal@vishal-OptiPlex-9020:~$ gcc --version
gcc (GCC) 4.7.3 20130411 (mspgcc dev 20120911)
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

vishal@vishal-OptiPlex-9020:~$ msp430-gcc --version
msp430-gcc (GCC) 4.7.3 20130411 (mspgcc dev 20120911)
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I am able to compile my code in Z1 mote Contiki. There is no any ROM error
but  I can't compile my code to sky mote.

greg-king5, 2019-03-28:
> Now, I am able to get the 4.7.3 for both.
> ```
> vishal@vishal-OptiPlex-9020:~$ gcc --version
> gcc (GCC) 4.7.3 20130411 (mspgcc dev 20120911)
> Copyright (C) 2012 Free Software Foundation, Inc.
> 
> vishal@vishal-OptiPlex-9020:~$ msp430-gcc --version
> msp430-gcc (GCC) 4.7.3 20130411 (mspgcc dev 20120911)
> Copyright (C) 2012 Free Software Foundation, Inc.
> ```

The `gcc` command must not run the same program that `msp430-gcc` runs!  `gcc` must run the Ubuntu system compilers.  Did you do the following advice?
>> The `update-alternatives` command added another alternative link that has a higher priority than the one that gcc should use.  You must remove the new link:
>> ```sh
>> sudo update-alternatives --remove gcc /opt/compiler/mspgcc-4.7.3/bin/msp430-gcc-4.7.3
>> ```

vishal623, 2019-03-29:
I have removed the link. Now the output is different.

vishal@vishal-OptiPlex-9020:~$ gcc --version
gcc (GCC) 4.7.3 20130411 (mspgcc dev 20120911)
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

vishal@vishal-OptiPlex-9020:~$ msp430-gcc --version
msp430-gcc (Ubuntu 5.5.0-12ubuntu1) 5.5.0 20171010
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

In my ~/.profile, only one path is there. Below is the line in my profile.

My ~/.profile file.

# ~/.profile: executed by the command interpreter for login shells.
# This file is not read by bash(1), if ~/.bash_profile or ~/.bash_login
# exists.
# see /usr/share/doc/bash/examples/startup-files for examples.
# the files are located in the bash-doc package.

# the default umask is set in /etc/profile; for setting the umask
# for ssh logins, install and configure the libpam-umask package.
#umask 022

# if running bash
if [ -n "$BASH_VERSION" ]; then
    # include .bashrc if it exists
    if [ -f "$HOME/.bashrc" ]; then
. "$HOME/.bashrc"
    fi
fi

# set PATH so it includes user's private bin if it exists
if [ -d "$HOME/bin" ] ; then
    PATH="$HOME/bin:$PATH"
fi

export PATH=$PATH:/opt/compilers/mspgcc-4.7.3/bin

My /etc/environment file

PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/opt/compilers/mspgcc-4.7.3/msp430/bin"
JAVA_HOME="/usr/lib/jvm/java-8-oracle"

greg-king5, 2019-03-30:
> I have removed the link. Now the output is different.
> ```sh
> vishal@vishal-OptiPlex-9020:~$ gcc --version
> gcc (GCC) 4.7.3 20130411 (mspgcc dev 20120911)
> Copyright (C) 2012 Free Software Foundation, Inc.
> 
> vishal@vishal-OptiPlex-9020:~$ msp430-gcc --version
> msp430-gcc (Ubuntu 5.5.0-12ubuntu1) 5.5.0 20171010
> Copyright (C) 2015 Free Software Foundation, Inc.
> ```
> In my "~/.profile", only one path is there. Below is the line in my profile:
> ```sh
> export PATH=$PATH:/opt/compilers/mspgcc-4.7.3/bin
> ```

OK, those arguments are backwards.  Swap them:
```sh
export PATH=/opt/compilers/mspgcc-4.7.3/bin:$PATH
```
By the way, have you tried using that "5.5.0" version of the msp430 compiler to build your mote programs?

> My "/etc/environment" file:
> 
> PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/opt/compilers/mspgcc-4.7.3/msp430/bin"

Apparently, that `gcc` command comes from the last path in that "PATH=" line.  That "bin/" directory is for old, old scripts that don't know about the "msp430-" prefix.  But, Contiki does know; it does use the prefix.  Therefore, that path shouldn't be in "/etc/environment".

vishal623, 2019-03-30:
No. It doesn't work for me.

I tried with msp430-5.5.0 it shows me the same error.  The error is
"msp430-gcc: error: unrecognized command line option ‘-mmcu=msp430f1611’".

When I tried to follow your instructions. it shows me the same.

vishal@vishal-OptiPlex-9020:~$ export
PATH=$PATH:/opt/compilers/mspgcc-4.7.3/binvishal@vishal-OptiPlex-9020:~$
msp430-gcc --version
msp430-gcc (Ubuntu 5.5.0-12ubuntu1) 5.5.0 20171010
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

vishal@vishal-OptiPlex-9020:~$ gcc --version
gcc (GCC) 4.7.3 20130411 (mspgcc dev 20120911)
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

vishal@vishal-OptiPlex-9020:~$ sudo update-alternatives --config gcc
There is only one alternative in link group gcc (providing
/usr/bin/msp430-gcc): /usr/bin/gcc-5
Nothing to configure.

vishal623, 2019-04-15:
I have reinstalled Ubuntu again.

I have two versions of msp430. (4.6.3 and 4.7.3). I have followed the
tutorial and set the path but it still shows me the version 4.6.3.

Following are the output of the given command:

vishal@vishal-OptiPlex-9020:~$ msp430-gcc --version
msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
Copyright (C) 2011 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

vishal@vishal-OptiPlex-9020:~$ cd /opt/compilers/mspgcc-4.7.3/
vishal@vishal-OptiPlex-9020:/opt/compilers/mspgcc-4.7.3$ ls
bin  include  lib  libexec  msp430  share

vishal@vishal-OptiPlex-9020:~$ cat /etc/environment
PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/opt/compilers/mspgcc-4.7.3/msp430/bin"
JAVA_HOME="/usr/lib/jvm/java-8-oracle"

vishal@vishal-OptiPlex-9020:~$ cat ~/.profile
# ~/.profile: executed by the command interpreter for login shells.
# This file is not read by bash(1), if ~/.bash_profile or ~/.bash_login
# exists.
# see /usr/share/doc/bash/examples/startup-files for examples.
# the files are located in the bash-doc package.

# the default umask is set in /etc/profile; for setting the umask
# for ssh logins, install and configure the libpam-umask package.
#umask 022

# if running bash
if [ -n "$BASH_VERSION" ]; then
    # include .bashrc if it exists
    if [ -f "$HOME/.bashrc" ]; then
. "$HOME/.bashrc"
    fi
fi

# set PATH so it includes user's private bin if it exists
if [ -d "$HOME/bin" ] ; then
    PATH="$HOME/bin:$PATH"
fi

export PATH=$PATH:/opt/compilers/mspgcc-4.7.3/bin

vishal@vishal-OptiPlex-9020:~$ gcc --version
gcc (Ubuntu 5.5.0-12ubuntu1) 5.5.0 20171010
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

If possible then anyone please guide me.

Regards,
Vishal

greg-king5, 2019-04-19:
> ```sh
> export PATH=$PATH:/opt/compilers/mspgcc-4.7.3/bin
> ```

That should be
```sh
export PATH=/opt/compilers/mspgcc-4.7.3/bin:$PATH
```

ali680, 2020-02-02:
how can change RPL in multi path in contiki os

munaalakhras, 2020-10-31:
> This is what i did (it works for me):
> 1- install ubuntu 14-04 (32bit)
> 2- download latest contiki 3.0 from sourceforge.net
> 2.1- extract contiki 3.0 files in "/home/"
> 3- Download the msp430-47 binary [...]

Thanks for this solution............since my ubuntu is 32 bit, and I attempted to install MSP430-gcc-4.7.3 but doesn't work because it only works for a 64-bit environment.      
At the last minute, I catch up with your comment before I decided to upgrade my ubuntu to 64 bit and I was about to lose all of my works in Contiki 2.7.
Again, Thanks for saving my time. 💯  

I recommend every one who has 32-bit ubuntu to use these steps literally

poornimagowda, 2022-06-29:
when i try to compile it will show the following error what to do?
> make hello-world.sky TARGET=sky 
make: msp430-gcc: Command not found
make: msp430-gcc: Command not found
fatal: Not a git repository: '../../.git'
mkdir obj_sky
  CC        ../../platform/sky/./contiki-sky-platform.c
make: msp430-gcc: Command not found
make: *** [obj_sky/contiki-sky-platform.o] Error 127
Process returned error code 2

poornimagowda, 2022-07-08:
I am a beginner to cooja I have installed Contiki 3.0 and tried to run examples hello-world. c...but it is showing a compilation error...please help me

> make hello-world.sky TARGET=sky 
make: msp430-gcc: Command not found
make: msp430-gcc: Command not found
fatal: Not a git repository: '../../.git'
  CC        ../../platform/sky/./contiki-sky-platform.c
make: msp430-gcc: Command not found
make: *** [obj_sky/contiki-sky-platform.o] Error 127
Process returned error code 2

hknaydin, 2022-07-23:
You have to install gcc-msp430 with the following command:

sudo apt-get install gcc-msp430