contiki-os/contiki #2281: Contiki nullrdc/nullmac duplicate packets?

fabricio-godoi, 2017-08-04:
I was testing some code in Cooja environment, and checking if all packets sent by clients were arriving in the respective order (given enough time). I checked that each packet came duplicate sometimes. I checked the RDC and MAC, and configured it with nullrdc and nullmac in contiki-conf.h. I copied the example "udp-ipv6" example and change some code, but is the same configuration.

The duplicate packets are coming in 10ms apart of the first packet. Anyone knows why the packets are coming duplicate? Am I forgetting to configure something?

By the way, I'm using Wismote with CC2520.

Thanks in advance, best regards!

msloth, 2017-09-04:
CSMA retransmissions due to the ACK is not received at the one sending the packet, as by design. Nullrdc should perform duplicate packet detection and drop it, if enabled, based on sequence numbers.

This question is better suited at eg stackoverflow.

fabricio-godoi, 2017-09-07:
Hello @msloth,
Thanks for the reply,

The nullrdc should discard packets based on the sequence number, but I was getting the same packet duplicated when checking the application. I'm waiting for the event of the tcpip_event, and checking if it's a uip_newdata() in the coordinator. Then if it's the application should perform as below:

Application method:
Any mote: send a specific packet with the packet number and source mote ID;
Coordinator: read all packets and check if is duplicated;

In almost all cases there is at least 2 packets with the same source and packet number (duplicated).
I don't know if the event is getting duplicated, or if there is some packet duplication in the routing.

I asked the same question at stackoverflow, but got no response, so I tried asking here for more information. Sorry if it's not intended to.

Thanks!