contiki-os/contiki #2295: a new implementation for pt

versaloon, 2017-08-11:
Implemented in https://github.com/versaloon/vsf, under vsf/framework/vsfsm.

#ifndef VSFSM_CFG_PT_GOTO
#define vsfsm_pt_begin(pt)			switch ((pt)->state) { case 0:
#define vsfsm_pt_entry(pt)			(pt)->state = __LINE__; case __LINE__:
#define vsfsm_pt_end(pt)			}
#else
#define vsfsm_pt_begin(pt)			if ((pt)->state) compiler_set_pc((pt)->state)
void vsfsm_pt_entry(struct vsfsm_pt_t *pt);
#define vsfsm_pt_end(pt)			
#endif

#ifdef VSFSM_CFG_PT_GOTO
void vsfsm_pt_entry(struct vsfsm_pt_t *pt)
{
	uint32_t lr;
	compiler_get_lr(lr);
	pt->state = lr;
}
#endif

#define compiler_set_pc(reg)			asm("MOV pc, %0" : :"r"(reg))
#define compiler_get_lr(reg)			asm("MOV %0, lr" : "=r"(reg))

This can implement similar algo like lc-addrlabels without GCC. But it needs 2 assembly for setting pc and getting lr. The code size is much better than switch style.

greg-king5, 2017-08-12:
> ```C
> #define compiler_set_pc(reg) asm("MOV pc, %0" : :"r"(reg))
> #define compiler_get_lr(reg) asm("MOV %0, lr" : "=r"(reg))
> ```

The opposite of "set the Program Counter" is "get the Program Counter", not "get the `lr` variable".

alexrayne, 2017-08-21:
lc-addrlabels alredy realises this pt-style via goto