contiki-os/contiki #2298: Secondary communication between nodes

Vinggui, 2017-08-18:
Hi guys,
   Inside of cooja, I need to create a secondary communications between sink nodes (only a few nodes would have internet access between all the nodes around), that would represent an Ethernet cable connection between them, so that they can synchronize and trade packets out of the main network being tested, like in a building monitoring or so.
   In real devices, I could do it, how to do something similar inside Cooja?

Thanks in advance.

msloth, 2017-09-04:
You could have a js-script running that basically accepts all serial output from one device and outputs it onto another device serial port (eg with a prefix and drop all other serial output), to simulate this.

This question is not an issue with Contiki and is better suited elsewhere, eg the developers mailing list or rather stackoverflow. See https://github.com/contiki-os/contiki/wiki#Community

Vinggui, 2017-09-06:
Thank you @msloth. I've being working on this and also found that your idea was a good solution.
I also agree that I may have used the wrong "channel" here. But thank you for your help!

closed 2017-09-06 by Vinggui