contiki-os/contiki #2300: Compile llsec with cc2538dk failed

laituandung, 2017-08-19:
Hello,

May I have your help with this issue.
I add following line to examples/cc2538dk/mqtt-demo/project-conf.h

/* nullrdc_driver: never switch the radio off */
#undef NETSTACK_CONF_RDC
#define NETSTACK_CONF_RDC     nullrdc_driver

/* Enable LLSEC */
#undef LLSEC802154_CONF_ENABLED
#define LLSEC802154_CONF_ENABLED          1
#undef NETSTACK_CONF_FRAMER
#define NETSTACK_CONF_FRAMER              noncoresec_framer
#undef NETSTACK_CONF_LLSEC
#define NETSTACK_CONF_LLSEC               noncoresec_driver
#undef NONCORESEC_CONF_SEC_LVL
#define NONCORESEC_CONF_SEC_LVL           0x07

#define NONCORESEC_CONF_KEY { 0x00 , 0x01 , 0x02 , 0x03 , \
                               0x04 , 0x05 , 0x06 , 0x07 , \
                               0x08 , 0x09 , 0x0A , 0x0B , \
                               0x0C , 0x0D , 0x0E , 0x0F }

Then compile with make command, however compilation failed with below output:
  CC        ../../../cpu/cc2538/./startup-gcc.c
  CC        mqtt-demo.c
  LD        mqtt-demo.elf
contiki-cc2538dk.a(sicslowpan.o): In function `output':
sicslowpan.c:(.text.output+0x50c): undefined reference to `noncoresec_framer'
contiki-cc2538dk.a(nullrdc.o): In function `send_one_packet':
nullrdc.c:(.text.send_one_packet+0x110): undefined reference to `noncoresec_framer'
contiki-cc2538dk.a(nullrdc.o): In function `packet_input':
nullrdc.c:(.text.packet_input+0x48): undefined reference to `noncoresec_framer'
collect2: error: ld returned 1 exit status
make: *** [mqtt-demo.elf] Error 1
rm mqtt-demo.co obj_cc2538dk/startup-gcc.o

Thanks,

laituandung, 2017-08-19:
Hello,

When I remove this code snippet. Compilation is successful.
#undef NETSTACK_CONF_FRAMER
#define NETSTACK_CONF_FRAMER noncoresec_framer

What is wrong with my definition?

slcmcu, 2019-02-25:
我用SM14Z2538开发板,需要联系我sun-lichao@163.com