contiki-os/contiki #2304: msp430-gcc upgrade

JunyangShi, 2017-08-22:
I can't compile the tsch-rpl for sky mote because of the limited ROM/RAM in this device. Can someone tell me how to upgrade msp430-gcc from 4.6.3 to 4.7.2 and solve the compilation problem?

alexrayne, 2017-08-23:
not eoung ROM can`t be solved by gcc upgrade.
i`ve build tsch-rpl minimalistic fro cc1310 - it acupies abut 50k ROM, and 12k RAM.
for 16bit cc430, it goes about half sizes - 25k ROM and 6k RAM. 

so looks that you have not enough RAM.
this can be reduced by:

1) turning on PT-threads manages event buffer less size,

2) set less TSCH buffers, and queues sizes.

look PR #2278 for a few more turnings on TSCH:
- there provided hard setup of mote style - coordinator, or slave. this a bit reduces code size.

alexrayne, 2017-08-23:
also current TSCH requires framer 802154. it eats more space too.
migrating to native rime chamelion frames - also cant help reduce RAM usage. but it need a lot of work.

JunyangShi, 2017-08-23:
Thank you so much. I'll try to tune the code in order to accommodate the size of ROM/RAM in sky mote.

alexrayne, 2017-08-23:
you`re have to strong redesign tsch memory usage;
1) turnoff tsch log - it ocupy 512bytes by default
2) tsch links pool - maybe it better to turn size for exact limits you need, or maybe better replace it by static allocated links
3) tsch queues - takes about 1-2k for, my platform. imho, not a problem to reduce it twice.
4) MEMB pool is not eficient storage - it use array of free flags, instead use bitmask. waste of memory not very much, but sufficient for your 4kB

knaggita, 2017-08-29:
These are some of the ways I found for tuning the code in order to accommodate the size of ROM/RAM in sky mote.
https://naggitakeziah.wordpress.com/2017/08/07/memory-optimization-in-z1-and-sky-motes/