contiki-os/contiki #2313: cc2538dk cannot use collect-view

yoyo890121, 2017-09-01:
I want to use `collect-view` to visualize my network topology so I tried `collect-view-shell` , but `collect-view` cannot find any connected motes. And I tried `motelist-linux`, the same as "No devices found".

I tried `rpl-border-router`  and used `tunslip6` connect to mote, it could work.

How do I get rid of the error? or Is there any other way to use `collect-view`?
Thanks.

lamnd09, 2017-09-04:
To my best knowledge, we only can use collect-view application with sky motes. It doesn't support cc2538dk at this time.

prodomo, 2017-11-15:
there is the guide of collect-view
http://anrg.usc.edu/contiki/index.php/Collect_View
could use command "make TARGET=cc2538dk" to produce .bin
P.S. it should have highest priority for step 4

yoyo890121, 2017-11-15:
Thanks for your sharing.
I tried `collect-view` on Windows and it can work.
I think Linux or VM is problem perhaps.

closed 2017-11-15 by yoyo890121

RaHeron95, 2019-07-18:
Hi @yoyo890121  I'm facing the same problem but using the cc2650 sensortag, could you explain to me how you could do it on windows?