contiki-os/contiki #2321: CC1310 network loss in medium-high loaded network

morphal, 2017-09-11:
Hello, 
I have faced an issue when nodes lose the network and refuse to join it unless they are rebooted. I have a custom cc1310-based device, but my colleagues with their own hardware design (cc1310-based) face the same issue.
The problem is reproduced when average network load is applied.  1 router + 3-5 nodes, udp ping-pong application (every node pings root once per second and root responds with pong) is enough to reproduce it.
One thing that I mentioned is that uart debug messages makes the situation worse. When all _core/net/rpl/rpl*.c_ files are set to DEBUG_FULL and uart speed is 115200, the issue is gonna be faced within 1-4 hours.
When uart is disabled the stability is higher, but within 24-48 hours I got 2 nodes (out of 5) to lose the network. 
When I increase debugging output I catch the issue within couple of minutes (I added debug output to MRHOF calculation function and got the issue in 45 seconds).
By losing the network I mean that the node deletes all routes and ignores further DIO messages.
I attach typical network losing log (see records before and after ‘Nullifying parents’ string)
Network setup: nullrdc, router is based on rpl-border-router example 
[nullifying_parents_logs.zip](https://github.com/contiki-os/contiki/files/1291651/nullifying_parents_logs.zip)

Can anyone help me please with eliminating this issue?
Thank you

niziak, 2017-09-11:
Please check #2161

vvzvlad, 2017-09-11:
I confirm the problem: with active communication on the network (we are working with OTA-update, where we need to send about 100 packets of 600 bytes), the devices stop working. In the worst case, the whole network stops working until all the devices are completely rebooted.
[log.txt](https://github.com/contiki-os/contiki/files/1292283/log.txt)

morphal, 2017-10-03:
@niziak, thanks. My issue is related to this indeed and your PR fixes it.