contiki-os/contiki #2322: RPL-tsch with wismote

sieben, 2017-09-12:
I've tried to compile the rpl-tsch example for the wismote/sky platform.
Here is the output:

> sieben @ sieben-inria in ~/workspace/contiki/examples/ipv6/rpl-tsch on git:upstream_master x [10:12:49] C:2
> $ make TARGET=wismote
>   CC        ../../../core/net/mac/tsch/tsch-slot-operation.c
> ../../../core/net/mac/tsch/tsch-slot-operation.c: In function ‘tsch_tx_slot’:
> ../../../core/net/mac/tsch/tsch-slot-operation.c:562:104: erreur: ‘RADIO_DELAY_BEFORE_TX’ undeclared (first use in this function)
> ../../../core/net/mac/tsch/tsch-slot-operation.c:562:104: note: each undeclared identifier is reported only once for each function it appears in
> ../../../core/net/mac/tsch/tsch-slot-operation.c:592:158: erreur: ‘RADIO_DELAY_BEFORE_RX’ undeclared (first use in this function)
> ../../../core/net/mac/tsch/tsch-slot-operation.c:597:202: erreur: ‘RADIO_DELAY_BEFORE_DETECT’ undeclared (first use in this function)
> ../../../core/net/mac/tsch/tsch-slot-operation.c: In function ‘tsch_rx_slot’:
> ../../../core/net/mac/tsch/tsch-slot-operation.c:762:98: erreur: ‘RADIO_DELAY_BEFORE_RX’ undeclared (first use in this function)
> ../../../core/net/mac/tsch/tsch-slot-operation.c:770:197: erreur: ‘RADIO_DELAY_BEFORE_DETECT’ undeclared (first use in this function)
> ../../../core/net/mac/tsch/tsch-slot-operation.c:876:126: erreur: ‘RADIO_DELAY_BEFORE_TX’ undeclared (first use in this function)
> ../../../Makefile.include:240 : la recette pour la cible « obj_wismote/tsch-slot-operation.o » a échouée
> make: *** [obj_wismote/tsch-slot-operation.o] Erreur 1
>

sieben, 2017-09-12:
@simonduq @yatch

yatch, 2017-09-12:
This is because `wismote` doesn't support TSCH. You can find supported platforms in https://github.com/contiki-os/contiki/blob/master/core/net/mac/tsch/README.md.

sieben, 2017-09-12:
Even with sky platform I get the following overflow error:

$ make TARGET=sky
  CC        node.c
  CC        ../../../platform/sky/./contiki-sky-main.c
  LD        node.sky
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: node.sky section
`.text' will not fit in region `rom'
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: node.sky section
`.bss' will not fit in region `ram'
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: section .vectors
loaded at [000000000000ffe0,000000000000ffff] overlaps section .text loaded
at [0000000000004000,00000000000126a1]
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: region `ram'
overflowed by 142 bytes
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: region `rom'
overflowed by 14412 bytes
/usr/lib/gcc/msp430/4.6.3/mmpy-16/libcrt0.a(_copy_data.o): In function
`__do_copy_data':
/build/buildd/gcc-msp430-4.6.3~mspgcc-20120406/./gcc-4.6.3/gcc/config/msp430/crt0.S:195:
relocation truncated to fit: R_MSP430_16_BYTE against symbol
`__data_load_start' defined in *ABS* section in node.sky
obj_sky/contiki-sky-main.o: In function `main':
/home/sieben/workspace/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:140:
relocation truncated to fit: R_MSP430_16 against symbol `memset' defined in
.text section in
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mmpy-16/libc.a(memset.o)
/home/sieben/workspace/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:142:
relocation truncated to fit: R_MSP430_16 against symbol `memcpy' defined in
.text section in
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mmpy-16/libc.a(memcpy.o)
/home/sieben/workspace/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:153:
relocation truncated to fit: R_MSP430_16 against symbol
`linkaddr_set_node_addr' defined in .text.linkaddr_set_node_addr section in
contiki-sky.a(linkaddr.o)
/home/sieben/workspace/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:154:
relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
/home/sieben/workspace/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:154:
relocation truncated to fit: R_MSP430_16 against symbol `printf' defined in
.text section in
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mmpy-16/libc.a(printf.o)
/home/sieben/workspace/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:156:
relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
/home/sieben/workspace/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:156:
relocation truncated to fit: R_MSP430_16 against symbol `printf' defined in
.text section in
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mmpy-16/libc.a(printf.o)
/home/sieben/workspace/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:158:
relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
/home/sieben/workspace/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:158:
relocation truncated to fit: R_MSP430_16 against symbol `printf' defined in
.text section in
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/lib/mmpy-16/libc.a(printf.o)
/home/sieben/workspace/contiki/examples/ipv6/rpl-tsch/../../../platform/sky/./contiki-sky-main.c:274:
additional relocation overflows omitted from the output
collect2: ld a retourné 1 code d'état d'exécution
../../../Makefile.include:280 : la recette pour la cible « node.sky » a
échouée
make: *** [node.sky] Erreur 1
rm node.co obj_sky/contiki-sky-main.o

yatch, 2017-09-12:
This is another issue, which should be relevant to https://github.com/contiki-os/contiki/issues/1968 and https://github.com/contiki-os/contiki/issues/1556; try a newer version of msp430-gcc ;-)

sieben, 2017-09-12:
Thanks a lot :)

closed 2017-09-12 by sieben