contiki-os/contiki #2324: hello-world example isn't working with Cooja Mote

ivan-alekhin, 2017-09-12:
Hello,
I'm trying to experiment with Cooja motes. I want to add new interfaces, but it appears I can't even run simulation for hello-world sample.
Can anyone help me please with this issue?
Thank you

### Environment

* Ubuntu 16.04, gcc-5.4.1, OpenJDK 1.8
* the latest contiki version from [master](https://github.com/contiki-os/contiki/commit/719f712307596c5f35c8f3f40824d9dfb110516b).

### Steps to reproduce

1. Run cooja.
```bash
cd tools/cooja
ant run
```
2. `<New Simulation> -> <Create>`
3. `<Motes> -> <Add motes> <Create new mote type> -> <Cooja Mote> -> select hello-world.c -> <Compile> -> <Create> -> Add 1 mote`
4. Start simulation.
5. Following stack trace appears:

```
java.util.NoSuchElementException
	at java.util.ArrayList$Itr.next(ArrayList.java:854)
	at org.contikios.cooja.radiomediums.AbstractRadioMedium.getActiveConnectionFrom(AbstractRadioMedium.java:211)
	at org.contikios.cooja.radiomediums.AbstractRadioMedium.access$500(AbstractRadioMedium.java:71)
	at org.contikios.cooja.radiomediums.AbstractRadioMedium$1.update(AbstractRadioMedium.java:382)
	at java.util.Observable.notifyObservers(Observable.java:159)
	at java.util.Observable.notifyObservers(Observable.java:115)
	at org.contikios.cooja.contikimote.interfaces.ContikiRadio.doActionsAfterTick(ContikiRadio.java:370)
	at org.contikios.cooja.MoteInterfaceHandler.doActiveActionsAfterTick(MoteInterfaceHandler.java:308)
	at org.contikios.cooja.contikimote.ContikiMote.execute(ContikiMote.java:152)
	at org.contikios.cooja.motes.AbstractWakeupMote$1.execute(AbstractWakeupMote.java:47)
	at org.contikios.cooja.Simulation.run(Simulation.java:281)
	at java.lang.Thread.run(Thread.java:748)
```

I've tried to fix that problem, and I've got another exception:

```
org.contikios.cooja.mote.memory.MemoryInterface$MoteMemoryException: Getting memory segment [0x54aab2e0,0x6daab2e1] failed: No section available
	at org.contikios.cooja.mote.memory.SectionMoteMemory.getMemorySegment(SectionMoteMemory.java:228)
	at org.contikios.cooja.mote.memory.Memory.getByteArray(Memory.java:165)
	at org.contikios.cooja.mote.memory.VarMemory.getByteArray(VarMemory.java:246)
	at org.contikios.cooja.contikimote.interfaces.ContikiRadio.doActionsAfterTick(ContikiRadio.java:336)
	at org.contikios.cooja.MoteInterfaceHandler.doActiveActionsAfterTick(MoteInterfaceHandler.java:308)
	at org.contikios.cooja.contikimote.ContikiMote.execute(ContikiMote.java:152)
	at org.contikios.cooja.motes.AbstractWakeupMote$1.execute(AbstractWakeupMote.java:47)
	at org.contikios.cooja.Simulation.run(Simulation.java:281)
	at java.lang.Thread.run(Thread.java:748)
```

I don't know how to fix that. I applied [this](https://github.com/contiki-os/contiki/pull/2033) patch, but it didn't help.

ivan-alekhin, 2017-09-20:
I've fixed the issue by adding `-Djava.compiler=NONE` to jvm args. I don't know why, but it looks like JIT influence on VarMemory (or JNI) behavior.

closed 2017-09-20 by ivan-alekhin

pengi, 2017-10-27:
I get the same error, with approximatly the same setup. (used OpenJDK, then Oracle JDK, both 1.8)

How did you fix the NoSuchElementException-problem?

ivan-alekhin, 2017-10-27:
@pengi,
It appears, the NoSuchElementException-problem wasn't a problem at all. It's just a symptom. In the end of my investigation, I've found that some of JVM objects can be corrupted by some reason. For instance, sometimes exceptions were appeared even in logger calls, and debugger showed really strange values of String Object fields. The root problem is related to JIT compiler or JVM internals, or native libraries. I have no idea why exactly it was happening, but I can suggest you following solutions:

1. Disable JIT compiler (`-Djava.compiler=NONE`). It's working, but it makes Cooja work on really slow rate.
2. Enable [CDS](https://docs.oracle.com/javase/8/docs/technotes/guides/vm/class-data-sharing.html) (`-Xshare:on`). That option also resolves the problem with unexpected JVM behavior, and this option doesn't disable JIT compilation, so Cooja will work with ordinary rate (or even faster that usually).

You can try to add one of these options to ant build file (`tools/cooja/build.xml`, target `run` or `run64`). For instance, `<jvmarg value="-Djava.compiler=NONE"/>` or `<jvmarg value="-Xshare:on"/>`.

mbaddeley, 2017-11-14:
`<jvmarg value="-Xshare:on"/>` worked a treat for me!

PauloSeixas, 2018-11-21:
<jvmarg value="-Xshare:on"/> worked a treat for me! when I installed the contiki in CentOS

reubenct, 2019-01-29:
@ivan-alekhin Thanks a lot!
Adding `<jvmarg value="-Xshare:on"/>` in target `run` made it work for me.

Strangely, my repository is missing `jni.h` altogether and I had to use the one available in OpenJDK 1.8. (The one from OpenJDK 1.7 didn't work out).