contiki-os/contiki #2333: Exception in thread "main" java.lang.UnsupportedClassVersionError: org/contikios/cooja/Cooja : Unsupported major.minor version 52.0

RamziFsm, 2017-09-20:
I installed Contiki 3.0 under Ubuntu 16.04 64 bits.But, i can't start cooja simulator. I am not able to understand what is the issue. How to resolve it? 
The result after building cooja is :

Buildfile: /home/ramzi/contiki/tools/cooja/build.xml

init:

compile:

copy configs:

jar_cooja:
      [jar] Building jar: /home/ramzi/contiki/tools/cooja/dist/cooja.jar

jar:

init:

compile:

jar:
      [jar] Building jar: /home/ramzi/contiki/tools/cooja/apps/mrm/lib/mrm.jar

init:

mspsim:

init:

compile:

jar:
      [jar] Building jar: /home/ramzi/contiki/tools/mspsim/mspsim.jar
     [copy] Copying 1 file to /home/ramzi/contiki/tools/cooja/apps/mspsim/lib

coffee:

init:

compile:
    [javac] Compiling 8 source files to /home/ramzi/contiki/tools/coffee-manager/build

configs:

jar:
      [jar] Building jar: /home/ramzi/contiki/tools/coffee-manager/coffee.jar
     [copy] Copying 1 file to /home/ramzi/contiki/tools/cooja/apps/mspsim/lib

compile:

copy:

jar:
      [jar] Building jar: /home/ramzi/contiki/tools/cooja/apps/mspsim/lib/cooja_mspsim.jar

init:

compile:

jar:
      [jar] Building jar: /home/ramzi/contiki/tools/cooja/apps/avrora/lib/cooja_avrora.jar

init:

clean:
   [delete] Deleting directory /home/ramzi/contiki/tools/cooja/apps/serial_socket/build

compile:
    [mkdir] Created dir: /home/ramzi/contiki/tools/cooja/apps/serial_socket/build
    [javac] Compiling 2 source files to /home/ramzi/contiki/tools/cooja/apps/serial_socket/build
[... 74 lines cut ...]
    [javac]   It is recommended that the compiler be upgraded.
    [javac] 12 warnings

jar:
      [jar] Building jar: /home/ramzi/contiki/tools/cooja/apps/powertracker/lib/powertracker.jar

run:
     [java] Exception in thread "main" java.lang.UnsupportedClassVersionError: org/contikios/cooja/Cooja : Unsupported major.minor version 52.0
     [java] 	at java.lang.ClassLoader.defineClass1(Native Method)
     [java] 	at java.lang.ClassLoader.defineClass(ClassLoader.java:803)
     [java] 	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
     [java] 	at java.net.URLClassLoader.defineClass(URLClassLoader.java:449)
     [java] 	at java.net.URLClassLoader.access$100(URLClassLoader.java:71)
     [java] 	at java.net.URLClassLoader$1.run(URLClassLoader.java:361)
     [java] 	at java.net.URLClassLoader$1.run(URLClassLoader.java:355)
     [java] 	at java.security.AccessController.doPrivileged(Native Method)
     [java] 	at java.net.URLClassLoader.findClass(URLClassLoader.java:354)
     [java] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:425)
     [java] 	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
     [java] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:358)
     [java] 	at sun.launcher.LauncherHelper.checkAndLoadMain(LauncherHelper.java:482)
     [java] Java Result: 1

BUILD SUCCESSFUL
Total time: 4 seconds

cherif-benali, 2021-02-12:
"Unsupported major.minor version 52.0 " i think it is a problem of JDK/JRE mismatch.
i had the same error because i programmed a java console program with JDK 1.8 and JRE  1.8 , in the other side my contiki 2.7 and cooja had 1.7 JDK/JRE , so i compile the project with 1.7 JDK and JRE and this problem is solved.

cherif-benali, 2021-02-12:
just look to the error : "javac] It is recommended that the compiler be upgraded." 
and " major version 52 is newer than 51, the highest major version supported by this compiler."