contiki-os/contiki #2335: running this RPL on contiki

hamidshabanipour, 2017-09-25:
hi 
how i can RUN this code with 8 motes on contiki? please guide me step by step.

hamidshabanipour, 2017-09-27:
how i can RUN this code  on contiki? please guide me step by step.

xioabudian, 2017-11-06:
for more information,refer to http://contiki-os.org/

hamidshabanipour, 2017-11-06:
i want to wirte my thesis about security evaluation of RPL protocol. i want to implement a complete RPL protocol in cooja simulator and an attack to RPL.then i want to evaluate behavior of nodes including sink node and sender nodes. can you help me about this subject?

xioabudian, 2017-11-06:
if you want cooja simulator,you could choose navtive platform, rpl-collect example.and you also update or change RPL protocol,however i do not know how to observe security experiment.But there are many differences between the actual situation and the simulation.I always do rearch on cc1310, not on simulator.

hamidshabanipour, 2017-11-06:
i saw a simulation of attacks in github. but i couldnt implemet this in my contiki.i just want a smple security evaluation. can you help me?