contiki-os/contiki #2341: mobility in Contiki 3.0

DavidAngc, 2017-10-05:
HI,
I have tried the mobility plugin on Contiki 2.7 and it is successful.
However, when i use it for Contiki 3.0, it does not work.
Can anyone advise how to implement the mobility plugin in Contiki 3.0?

Regards,

Gotacotrim, 2017-10-24:
Hi David,

First thing to do is change the libraries on Mobility.java if you didn't do yet
(old)   import se.sics.cooja.ClassDescription;
(new) import org.contikios.cooja.ClassDescription;

Then you will get some problems with the line 
private MessageList log = new MessageList();

The MessageList library has changed to MessageListUI, so you will need to update the code.

I hope this could help you.

regards,

avitzavi, 2018-01-08:
I also had to update the path to cooja.jar in the build.xml 
BUT:
Even after changing the libraries (all of them, not just .ClassDescription) in Mobility.java and doing a find/replace of MessageList to MessageListUI, I am still getting errors when trying to build the Mobiity package for the first time.

avitzavi, 2018-01-09:
I got it to successfully build with the following changes to Mobility.java:
1) changed all mentions of "import se.sics.cooja." to "import org.contikios.cooja."
2) changed "import org.contikios.cooja.GUI;" to "import org.contikios.cooja.Cooja;"
3) changed ~2 mentions of getGUI() to getCooja
4) changed ~3 mentions of "GUI" (on those in all caps) to "Cooja"

I did not change MessageList to MessageListUI in any location. 

Let me know if this sounds right!

JayanthEM, 2018-06-30:
Hi,
The mobility plugin is working for me in contiki 3.0.
Refer this blog http://contikiemj.blogspot.com.

JihenHlel, 2018-08-09:
@avitzavi  and @JayanthEM  i did the same steps you followed but i didn't get mobility in tools, 
Any propositions???

JayanthEM, 2018-08-10:
In the mobility plugin check the cooja.config file. If the directory path is like se.sics.cooja... change it to org.contikios.cooja...

JihenHlel, 2018-08-10:
it's ok @JayanthEM  thank you i solved the problem, this link may help others: http://miracbektas.com/?p=123 , @JayanthEM  you didn't sign anything about config cooja file in your blog you have to updated it :)

ayoub-root, 2018-09-20:
follow this link but you need to chnage something in mobilty.java 
replacethe line  messagelist log = new messagelist(); by  messagelistui  log = new messagelistui()

faroouq, 2019-04-03:
After step in  "ayoub-root commented on Sep 20, 2018" above I run **_sudo ant clean_** in the mobility folder to clean previous compilation. Then I run **_sudo ant jar_** to make new compilation.

It runs. 
Thanks all.

tarimislam, 2019-05-08:
I am facing problem to start onus in Contiki. Problem is given above.
[image]
 Any solutions

nilampradhan2, 2022-10-15:
I am trying to build mobility in contiki-ng. it is giving me following error. Why is it showing /../../tools/tools/.../ in path. please help

[java] WARN [AWT-EventQueue-0] (ProjectConfig.java:240) - Trying to import non-existant project configuration: /home/user/contiki-ng/tools/tools/cooja/apps/mobility/cooja.config
[java] FATAL [AWT-EventQueue-0] (Cooja.java:437) - Error when loading extensions: Error when loading extension: Project directory does not exist: /home/user/contiki-ng/tools/tools/cooja/apps/mobility
[java] org.contikios.cooja.Cooja$ParseProjectsException: Error when loading extension: Project directory does not exist: /home/user/contiki-ng/tools/tools/cooja/apps/mobility

TiagoDemay, 2022-11-01:
> Estou tentando construir mobilidade em contiki-ng. está me dando o seguinte erro. Por que está mostrando /../../tools/tools/.../ no caminho. por favor ajude
> 
> [java] WARN [AWT-EventQueue-0] (ProjectConfig.java:240) - Tentando importar configuração de projeto inexistente: /home/user/contiki-ng/tools/tools/cooja/apps/mobility/cooja.config [java] FATAL [AWT-EventQueue-0] (Cooja.java:437) - Erro ao carregar extensões: Erro ao carregar extensão: diretório do projeto não existe: /home/user/contiki-ng/tools/tools/cooja/ apps/mobility [java] org.contikios.cooja.Cooja$ParseProjectsException: Erro ao carregar a extensão: o diretório do projeto não existe: /home/user/contiki-ng/tools/tools/cooja/apps/mobility

@nilampradhan2,  I believe the PATH is wrong, there are two "tools" in yours.