contiki-os/contiki #2355: CC1350: question on 'rime-tsch' example

aristanemi, 2017-10-12:
Hello,
I am trying to use rime+TSCH on CC1350 sensortag. 
1. make TARGET=srf06-cc26xx BOARD=sensortag/cc1350 at 
/home/ga83jok/contiki/examples/rime-tsch/
2. I am getting following error,
```
Starting Contiki-3.x-3336-g1364e98
With DriverLib v0.47020
TI CC1350 SensorTag
IEEE 802.15.4: Yes, Sub-GHz: Yes, BLE: Yes, Prop: Yes
 Net: Rime
 MAC: TSCH
 RDC: nordc
TSCH:! radio does not support getting RADIO_PARAM_RX_MODE. Abort `init.`
```
Can anyone help me with this error. Is TSCH still not ported to CC1350.

Regards,
Abhishek

chenek, 2017-10-23:
As I know, TSCH is still not ported to CC13xx

arurke, 2017-10-23:
You could checkout the string of PRs from @alexrayne, maybe most notably #2273