contiki-os/contiki #2357: cc2650: strange pattern of resets

adamrenner, 2017-10-13:
I have found a strange problem with the cc2650. I am also using the TSCH along with a Rime mesh.
I have run the software on Cooja motes and other hardware I cannot repeat it elsewhere, so I think that the hardware play a role in the bug.

When I run the motes, I have them report a node up time once per second along with other data using clock_seconds().

I would sometimes get seemingly random resets but after sometime, I began to notice that all of the resets seemed to happen at regular intervals. The last uptime reported would always be one less than 540*n. Where n was any positive integer. So they might reset at 540 seconds, 1180 seconds, 1720 seconds, etc...

540 seconds happens to be 9 minutes. Sometimes the system will run for hours, but when it resets, it will always be at an interval of 540 seconds.

I have been chasing this bug for awhile and cannot seem to locate the root cause and was hoping that the 540 second interval might be something that stands out to someone else.

atiselsts, 2017-10-13:
That's probably related to the RAT overflow timer, which is scheduled to be checked once 9 minutes.

Are you using the latest version? #2138 fixes some of the problems in CC2650 IEEE mode radio driver. Try enabling logging in `ieee-mode.c, possibly adding mode printfs to see what exactly happens before the reboot.

On a second thought, the problem also could be related to #2140, a pull request that fixes a critical bug in TSCH that sometimes cause watchdog resets, in particular on CC2650. It's not merged yet.

adamrenner, 2017-10-17:
Thanks!
My version has pulled in changes up to 20 days ago, so #2138 is included, but I will look deeper into ieee-mode.c logging.

I will also run a test against #2140.

Will let you know what I find.