contiki-os/contiki #2358: PAN ID in prop mode(cc1310) and ieee mode?

vvzvlad, 2017-10-13:
I'm trying to use NETSTACK_RADIO.set_value (RADIO_PARAM_PAN_ID, panid); to set the network identifier, but this works on the CC2650, but does not work when I collect the code for CC1310. I checked and understood that in the case of 1310 "prop mode" is used, and with 2650 "ieee mode".

1) Can I use pan id with the network on cc1310?
2) Can I use ieee mode for the network on cc1310?
3) What is the difference between ieee mode and prop mode?

alexrayne, 2017-10-23:
ieee mode on 1310 can`t use advanced radio commands. 
thouse commands allows use user-specified frame header. And, most important, allows use wide range on radio chanels, vs a just a couple for ieee mode.
tsch, when use prop-mode, need to build ieee81xxxx frame header by software.

xioabudian, 2017-11-06:
question 2:no,cc1310 driver is prop-mode,not ieee mode.you can see compile and link process
question 3:ieee-mode is on cc2650,2.4GHz.prop-mode is on cc1310,Sub-GHz.and the drivers are so much difference.

xioabudian, 2017-11-06:
add a little more,PANID has nothing to do with driver.maybe you could achieve it on cc1310 platform

xioabudian, 2017-11-06:
add a little more,as far as i know,PANID has nothing to do with driver.maybe you could achieve it on cc1310 platform